markqvist/Sideband #2: please remember increasing versionCode

IzzySoft, 2022-09-28:
Despite of multiple releases already having been published, the latest APK still identifies as `versionCode: 1`. This makes automated updates hard, as Android uses `versionCode` to tell versions apart – so it considers your latest release being the same version as the initival one and doesn't offer it for update (`versionName` is just for humans to read).

May I hence suggest you increase `versionCode' with each new release?

markqvist, 2022-09-29:
This is a complete oversight on my part, not being very used to the structure of Android packaging. Thanks a lot for bringing it up! I'll make sure to increase the versionCode for each new release from this point on.

closed 2022-09-29 by markqvist

IzzySoft, 2022-09-30:
Thanks a lot! That accomplished, I've just added your app to my repo – it will [show up here](https://apt.izzysoft.de/fdroid/index/apk/io.unsigned.sideband) with the next sync around 6 pm UTC today. Updates are pulles within 24h of your tagging a new release and attaching the APK to it. Be welcome to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) if you wish to link there :smiley: 

PS: I gladly provide you with the metadata used for the listing via PR here, so you come into full control of how your app is presented – just let me know if you want it.

IzzySoft, 2022-10-04:
> That is awesome, really appreciate you adding it!

Glad you like it :smiley:

> I am completely happy for you to manage presentation in the repo, I have plenty to do with everything else ;)

Fine with me. But in that case I have to rely on your notification should changes be necessary (like updates to the descriptions, screenshots etc).

> If you don't mind, can you sync it to [the latest release](https://github.com/markqvist/Sideband/releases/tag/0.1.9) I just put out now?

That should happen automatically. My updater is running regularly (every 24h) to check for new tags with a fitting APK attached to it and pulls it then (along with the metadata, if they are in the repo as described). That said: The link you gave gives a 404: there is no tag `0.1.9`, so nothing I can pull.

> I have included more screenshots and presentation resources that you can pick and choose from

Wonderful! Those would be part of the said metadata. An established format for that is Fastlane (see my [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688)) which my updater can deal with automatically – so if you could place the screenshots you want added into `fastlane/metadata/android/en-US/images/phoneScreenshots` (usually from the repository root, where F-Droid.org would expect it – though with my updater the "starting place" is configurable) I'd tell my updater to check them with each new release. For now I've pulled Screen 1 to 6 manually (in addition to screenshot 1-5), which is what I'd suggest to be used. Will become effective with the next sync today around 6 pm UTC.

markqvist, 2022-10-04:
Sorry, I had to yank the 0.1.9 relase because of a bug. There is a new one up now though, sorry for the confusion!

> That should happen automatically

Ah, that is nice! Then there's nothing else to do than wait :)

> Those would be part of the said metadata. An established format for that is Fastlane ...

That is brilliant, I will get this set up as soon as I can. In that case, will it pull the updated metadata as soon as it detects it, or will it need a version bump to trigger it?

markqvist, 2022-10-04:
Yeah, I see now ;) I am going to get it set up like that. Looking forward to the sync, but pretty cool to see Sideband in F-Droid via your repo already. Nice work!

IzzySoft, 2022-10-05:
Great! Please give me a ping when it's there and I should check, so I can set up the config on my end to pull from yours.

IzzySoft, 2023-03-31:
I've just been notified you've changed the license to a non-FOSS one (essentially, the "NC" makes it non-FOSS). Would you consider a different license, maybe? Else I'm not sure if I can keep it in my repo, which is dedicated to FOSS.

markqvist, 2023-04-02:
Yes, I will consider that. I will be revamping a few of the licenses soon. I recently had to temporarily change licenses on a few projects, for reasons discussed elsewhere.

IzzySoft, 2023-04-02:
Thank you! Can you send me a ping (e.g. via a comment here in this issue) when the license is decided, so I adjust then properly?

markqvist, 2023-04-05:
Will do!

IzzySoft, 2023-05-07:
Due-date popped up with the issue in my repo. Still no news? The license here still is the "no-F/LOSS" one. I'll shift the due-date one more time, but then a decision should stand. Hope it can be done until then?

markqvist, 2023-05-09:
I hope so too, but I currently can't easily recompile everything, due to some annoying dependency issues in Kivy, KivyMD, SDL and python-for-android, so I am waiting for bugfixes to be upstreamed and released in those projects, which hopefully should happen soon.

IzzySoft, 2023-05-09:
Adding a license file (and/or a corresponding note in the Readme) should not require a recompile I guess? :wink:

markqvist, 2023-05-09:
It will require a recompile to get the license notes and license header into the about section of the program and such, and unfortunately I don't have time for it right now.

IzzySoft, 2023-05-09:
That part is clear of course. But having it presented in the repo here already would be possible, right? Maybe along with a note that the references in the app will be updated soon™ as well?

IzzySoft, 2023-06-03:
@markqvist any chance? The due-date popped up in my issue tracker again. So any ETA when you will have at least the license in the repo re-adjusted? I'll shift the due date now for a last time. If the license file hasn't been replaced then, I'll have to disable/remove your app from my repo until it's fixed, sorry.

IzzySoft, 2023-07-07:
You still did not manage to replace the license back after 3 months now – and since May you're not even responding anymore. Is there any chance left you'll assign this a libre license again?

IzzySoft, 2024-01-31:
It's been over a year now. As my repo is reserved for FOSS apps using [FOSS licenses](https://spdx.org/licenses/) (i.e. those approved by OSI/FSF), I will now have to remove your app from there. Be welcome to ping me should you have changed to such a license, and the listing can be re-established.

commit 48d7dda2 (2022-09-29): Specified meaningful versionCode for Android. Closes #2.