markqvist/Sideband #3: Sideband not working on Motorola One 5G Ace

OtisByron, 2022-10-07:
Since v0.1.9 I am unable to run Sideband on my Motorola One 5G Ace. The app side-loads successfully, but when I try to open the app it stops on the "Please Wait, Substantiating Reticulum" screen. The "Sideband Active" notification also goes away after about 3 seconds.

I have tried reinstalling the app several times, as well as clearing storage and cache several times. I have had this issue with versions 0.1.9, 0.2.0, and 0.2.1.

I am able to run v0.2.1 on my Samsung Galaxy S9, so it appears to be an issue with my Motorola phone.

Has anyone else experienced this issue? Any ideas on how to troubleshoot this further?

markqvist, 2022-10-07:
Since that was when the app was changed to run in "split-brain mode" on Android, ie. run all the Reticulum and LXMF stuff in a background service, and the app UI in the foreground, it probably has something to do with that. It sounds like the service tries to start up, but then dies, leaving the app waiting forever for it to come up.

If you have the time, and a computer with `adb` installed (the Android debug bridge), you can help me out a lot by capturing the log output at startup. I have just compiled an APK with full debug log output, that you can download here:

https://reticulum.network/dl/sideband-0.2.1-arm64-v8a-debug.apk

You can capture the log output by running the command `adb logcat | grep "sidebandservice"` (it's going to be a lot of text).

If no apparent errors crop up there, you can get the full output for both the service and the foreground app with: `adb logcat | grep "python\|sidebandservice"`.

If you can send me that log output, I should be able to debug it and have a fix for you :)

OtisByron, 2022-10-07:
Thanks for the support Mark. I'm traveling at the moment, but I plan to work through this and capture the log output as soon as I can.

markqvist, 2022-10-07:
That is great, thank you for taking the time. Please just let me know whenever you have the time for it.

markqvist, 2022-10-15:
This *may* have been fixed in the latest release: https://github.com/markqvist/Sideband/releases/tag/0.2.4

Could you give it a try when you have time?

OtisByron, 2022-10-16:
Unfortunately the latest release linked above does not fix the issue. I was about to get adb installed today and captured the following output:

[sideband_adb_logcat.txt](https://github.com/markqvist/Sideband/files/9795252/sideband_adb_logcat.txt)

markqvist, 2022-10-19:
Thanks. It looks like that output is from the release build of version 0.2.4, and unfortunately it hasn't logged any cause of the problem. Could you try with this build, that has more debug logging enabled?

https://github.com/markqvist/Sideband/files/9819617/sideband-0.2.5-arm64-v8a-debug.zip

If you can give me the `adb logcat` output of the above build, I should be able to fix it!

DarkLinkXXXX, 2022-11-25:
I'm experiencing this now, haven't tried running adb yet but I have a hunch this is from something being killed for not having enough memory, as my Motorola G Power has only 3G of ram and I keep trying to run an impossible amount of stuff in the background.

markqvist, 2022-11-25:
Thanks for reporting it @DarkLinkXXXX. It sounds like it could be some kind of memory management issue. Can you try disabling battery optimisations for Sideband in the system settings?

Sideband launches a background service when it starts (the service runs the Reticulum protocol layers), and if that gets killed it will not be able to start up.

The current release, version 0.4.0, does not log a lot of debug output by default, but you might still be able to see errors. I will have the option to enable full debug log output, and a way to export it without adb in the next version :)

OtisByron, 2023-01-22:
I apologize for not getting back to this sooner. Here is the adb logcat output for the 0.2.5 debug version you posted. It does look like there is an exception.

[sideband_adb_logcat_20230122.txt](https://github.com/markqvist/Sideband/files/10475099/sideband_adb_logcat_20230122.txt)

simplyequipped, 2023-02-10:
A few solutions to this exception are mentioned in [this post](https://stackoverflow.com/questions/41986851/caused-by-java-lang-noclassdeffounderror-class-not-found-using-the-boot-class) in case this is helpful.

markqvist, 2024-01-03:
Assuming this is fixed with the earlier changes to the way Sideband handles Android API calls and permissions. If not, feel free to provide a new log dump from the latest Sideband version (which also includes better error logging).

closed 2024-01-03 by markqvist