markqvist/Sideband #5: Copy message text strange delay.

SebastianObi, 2022-10-09:
Hi Mark,

I tested it again. It's not that it doesn't work.
It is only a little bit strange behaviour:
When I click on copy and after this direct paste it there is only the beginning of the text. When I wait some seconds then I have the whole text.

It looks like the copy function is very slow and need some time to send all characters to the clipboard.

I think it's an android issue and has nothing to do with your code.

I used Sideband 0.2.1beta on Android 12.

markqvist, 2022-10-13:
It is pretty weird... I have tried this on five different Android devices now, and I am not able to reproduce it on any of them. Could it be that in only happens on the specific device you are using? You are welcome to share what device you are seeing this on.

I have experienced slow clipboard behaviour before on Android when system extensions are enabled to display notifications whenever an app interacts with the clipboard (as a security feature).

SebastianObi, 2022-10-15:
It seems that only that phone is not working. I just tested on another android phone. There everything is OK. Strange :)

Issue:
Samsung Galaxy S20FE
Andoid 12
Kernel-Version 4.19.113-24784831

OK:
Samsung Note 9
Andoid 10
Kernel-Version 4.9.118-24343300

If you want you can close the issue.

markqvist, 2022-10-15:
Thanks for the info. I think I will close it for now, since we can't get any further right now really, and it is quite a minor thing.

If anyone experiences this in the future, please comment here with info so we can investigate it further.

closed 2022-10-15 by markqvist