markqvist/Sideband #6: Message sync wrong message order (Timestamp)

SebastianObi, 2022-10-12:
Hi Mark, 
I have discovered the following small unsightly behavior. When downloading multiple messages during a sync, the order is not correct. They should be added sorted according to the send timestamp.

This is not a real bug but just some cosmetics and improvement :)

I used Sideband 0.2.2

markqvist, 2022-10-13:
Hmm, this is an interesting one. The decryption happens so late in delivery process, that the LXMF message router actually doesn't have access to the message timestamps (they are part of the encrypted message data).

It is not until the LXMs are handed over to the final delivery destination, that we are able to resolve the timestamps, and then, they will of course "already have been delivered" and have their internal delivery timestamp set, which is what get's sorted on by default. The logic behind this is that you want to see the latest *delivered* messages most prominently in your message stream, even though they might have been sent a long time ago.

Due to the mix between direct and asyncronous communication that LXMF supports, you might receive a message sent a few days ago via propagation nodes. If this message just "disappears" into the message stream past, the user may have a hard time finding it.

The way I "solved" this is Nomad Network was to just allow the user to sort each conversation based on message timestamps or delivery order, using a keyboard shortcut.

Maybe we could do something similar in Sideband. What do you think? I realise this is related to your group conversation tool, so maybe a switch per-conversation for sorting or something like that would be useful.

Also, just as a side-note to anyone else reading this, built-in group conversation support is coming to LXMF in the pretty near future :)

SebastianObi, 2022-10-15:
Hello Mark,

I didn't mean that the new messages are completely sorted with the existing messages. As you said this would be very bad after all. I was thinking of all the new messages received at once just being sorted among themselves and then appended at the bottom.

A manual sorting like at nomadnet would of course also be very handy.

Good to hear that the group feature is being worked on. :)

My solution is a workaround for this. As a server side group also with other functions.

closed 2022-10-25 by markqvist