markqvist/Sideband #7: Transport instance not working? - Sideband β 0.2.4

SebastianObi, 2022-10-15:
I have just tested the new feture. However, it does not seem to work for me.

Here is the scenario:
Android-A: TCP testnet + RNode (Enable Trabsport)
Android-B: RNode

The traffic/comunication is not routed between the TCP testnet and Android-B.

Android-A reaches the testnet and Android-B

markqvist, 2022-10-15:
Wow, you are fast.

Just for good measure, could you please retry with the APK I uploaded just now? The first one had a bug, so please reinstall.

Also, remember you must shut down and restart Sideband for transport to get started. You should be able to see it as active in the connectivity status dialog.

SebastianObi, 2022-10-15:
I know that i need to shutdown Sideband ;)

Now I have tried it again. It does't work :(
I tested announces and messages.

The first thing I did was update the app.
As a 2nd attempt completely uninstalled and reinstalled.

This test was on my phone with android version 10.

SebastianObi, 2022-10-15:
On the Android version 12 also don't work.
It looks like it may be a general issue withe the transport on Andoid.

markqvist, 2022-10-15:
Yeah sorry, sometimes it's just the obvious things that gets overlooked ;)

I need to get to the bottom of it then. It was honestly a bit of a last-minute addition, so I only tested it on Android 13 so far. I'll get out some more devices and see what is going on.

markqvist, 2022-10-15:
I found it. I will have an update APK with working Transport functionality in a moment.

markqvist, 2022-10-15:
I reuploaded the APK with a fix, please give it a try :)

markqvist, 2022-10-15:
I am closing this now since I am pretty certain it is fixed, but feel free to reopen if it still doesn't work in your end.

And thanks a lot for catching it so quickly and reporting it! Really appreciate it!

closed 2022-10-15 by markqvist

SebastianObi, 2022-10-18:
Hi, 
I I had done another test. In this scenario the routing did not work. It seems not to route between RNode and Local interface. Between RNode and TCP it had worked.

I had activated a WiFi hotspot and connected a 2nd phone to this WiFi. Then the traffic was not routed to LoRa.

Could it be that something is still not quite working?

markqvist, 2022-10-19:
Did you activate the wifi hotspot after launching Sideband? In that case, the OS will bring up a new kernel networking device (probably something like *wlan1* or *wlan2*) for the hotspot, but Sideband is not going to automatically adopt it, if it appears *after* the app has started.

I might be asking for something obvious here, but I am doing so since I can't replicate the problem. Here it is working perfectly with an RNode connected to one phone and that phone also creating a wifi hotspot for other phones to connect to. All Sideband instances can route traffic through the RNode.

There is also the chance that some of the small fixes I have been working on in the meantime has also fixed this issue, and that is why I can't replicate it. You are welcome to give this debug build a try, I will attach it to this post.
[sideband-0.2.5-arm64-v8a-debug.zip](https://github.com/markqvist/Sideband/files/9819617/sideband-0.2.5-arm64-v8a-debug.zip)

SebastianObi, 2022-10-19:
I just tested it with the debug version. It works :)))
Thanks a lot.
Yes you have to make sure to start the hotspot first.