markqvist/Sideband #8: Shutdoown don't work (App is not completely terminated) - Sideband β 0.2.4

SebastianObi, 2022-10-15:
When you click on "Shutdown" in the menu, the Extit Screen "Reticulum Dissolving" is displayed. Then also closed and the app is no longer displayed. So it looks like Sideband is really finished. However, if you switch to the view of all running apps. Sideband is still displayed. If you select the app there, Sideband starts again.

This is probably only a small, insignificant issue. Nevertheless, I have created an issue for it.

Tested on Samsung S20FE Android 12 & Samsung Note 9 Android 10.

markqvist, 2022-10-15:
Thanks, I am seeing the same behaviour, but I am actually not sure if it is possible to remove the app from the activity overview, or if it is only possible by the user "swiping" it away. The app *is* actually terminated, but the last frame of the program is still left in the activity overview switcher.

There might be an Android API to remove the app from the overview, I will see if I can find something. If anyone knows, please don't hesitate to tell me ;)

markqvist, 2022-10-15:
Ok, I found a number of different approaches in the Android APIs, ranging from hacky to possibly fine. It seems that there isn't really any "official" way to do this, but that it can be done by bending the APIs a little.

I will experiment with it for the next release and see if there is a solid way to implement it. It's a minor thing, but I agree it would be very nice for it to have that behaviour on exit.

marcdw1289, 2022-10-16:
Yeah. If you notice that most web browsers today completely disappear  and not visible in overview/recents when using their exit/quit menu entries.
Maybe have a look see at how they accomplish that.

markqvist, 2022-10-19:
Got it. Found some different methods that may work, gonna see if it can be added with relative ease.

markqvist, 2022-10-19:
It should work now, thanks for reporting it!

commit b432646c (2022-10-19): Improved exit behaviour on Android. Closes #8.
closed 2022-10-19 by markqvist