markqvist/Sideband #9: Support armeabi-v7a android phones

Cinorid, 2022-10-16:


markqvist, 2022-10-19:
Hi @Cinorid 

Could you provide some more detail or context to this request?

I am interested to hear what kind of devices you want to run Sideband on, that needs API level 27 and the v7a ABI?

markqvist, 2022-10-19:
Also, what specific API levels are you wanting support for? Just "lower than 27" is a pretty wide target ;)

markqvist, 2022-10-19:
Thanks @Cinorid, got you. 

Here is a debug build for you targeted at Min API level 24 and armeabi-v7a. Can you please try it out? Unfortunately I don't even own any Android devices with a 32-bit SoC, so I have not even tried installing it myself. Don't know if it will even work, but please give it a try and let me know how it works!

[sideband-0.2.5-armeabi-v7a-debug.zip](https://github.com/markqvist/Sideband/files/9820080/sideband-0.2.5-armeabi-v7a-debug.zip)

Cinorid, 2022-10-19:
> 

sure. I'm Testing :)

Cinorid, 2022-10-19:
@markqvist 
Unfortunately it stuck in loading screen
[image]

markqvist, 2022-10-19:
Can you give me output of `adb logcat | grep "python\|sidebandservice"` when starting the app?

Cinorid, 2022-10-19:
> Can you give me output of `adb logcat | grep "python\|sidebandservice"` when starting the app?

```
10-19 15:38:02.097  5693  5762 I python  : Initializing Python for Android
10-19 15:38:02.097  5693  5762 I python  : Setting additional env vars from p4a_env_vars.txt
10-19 15:38:02.098  5693  5762 I python  : Changing directory to the one provided by ANDROID_ARGUMENT
10-19 15:38:02.098  5693  5762 I python  : /data/user/0/io.unsigned.sideband/files/app
10-19 15:38:02.102  5693  5762 I python  : Preparing to initialize python
10-19 15:38:02.102  5693  5762 I python  : _python_bundle dir exists
10-19 15:38:02.103  5693  5762 I python  : calculated paths to be...
10-19 15:38:02.103  5693  5762 I python  : /data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip:/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules
10-19 15:38:02.121  5693  5762 I python  : set wchar paths...
10-19 15:38:02.302  5693  5762 I python  : Initialized python
10-19 15:38:02.302  5693  5762 I python  : AND: Init threads
10-19 15:38:02.304  5693  5762 I python  : testing python print redirection
10-19 15:38:02.306  5693  5762 I python  : Android path ['.', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/site-packages']
10-19 15:38:02.307  5693  5762 I python  : os.environ is environ({'PATH': '/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin:/system/vendor/bin:/vendor/xbin:/system/vendor/xbin:/product/bin:/product/xbin', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'BOOTCLASSPATH': '/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/hwEmui.jar:/system/framework/hwTelephony-common.jar:/system/framework/hwframework.jar:/system/framework/org.simalliance.openmobileapi.jar:/system/framework/org.ifaa.android.manager.jar:/system/framework/hwaps.jar:/system/framework/hwcustEmui.jar:/system/framework/hwcustTelephony-common.jar:/system/framework/hwcustframework.jar:/system/framework/servicehost.jar', 'SYSTEMSERVERCLASSPATH': '/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/hwServices.jar:/system/framework/hwWifi-service.jar:/system/framework/hwcustServices.jar:/system/framework/hwcustwifi-service.jar', 'OEM_ROOT': '/hw_oem', 'CUST_POLICY_DIRS': '/system/emui/base:/system/emui/oversea:/system/emui/lite:/system/global:/vendor/etc:/product/etc:/hw_oem:/cust/hw/normal:/data/cust:/version/region_comm/oversea:/cust_spec', 'ANDROID_SOCKET_zygote': '14', 'ANDROID_ENTRYPOINT': 'main.pyc', 'ANDROID_ARGUMENT': '/data/user/0/io.unsigned.sideband/files/app', 'ANDROID_APP_PATH': '/data/user/0/io.unsigned.sideband/files/app', 'ANDROID_PRIVATE': '/data/user/0/io.unsigned.sideband/files', 'ANDROID_UNPACK': '/data/user/0/io.unsigned.sideband/files/app', 'PYTHONHOME': '/data/user/0/io.unsigned.sideband/files/app', 'PYTHONPATH': '/data/user/0/io.unsigned.sideband/files/app:/data/user/0/io.unsigned.sideband/files/app/lib', 'PYTHONOPTIMIZE': '2', 'P4A_BOOTSTRAP': 'SDL2', 'PYTHON_NAME': 'python', 'P4A_IS_WINDOWED': 'True', 'P4A_ORIENTATION': 'sensor', 'P4A_NUMERIC_VERSION': '20221018', 'P4A_MINSDK': '24', 'LC_CTYPE': 'C.UTF-8'})
10-19 15:38:02.307  5693  5762 I python  : Android kivy bootstrap done. __name__ is __main__
10-19 15:38:02.307  5693  5762 I python  : AND: Ran string
10-19 15:38:02.307  5693  5762 I python  : Run user program, change dir and execute entrypoint
10-19 15:38:03.093  5693  5762 I python  : [WARNING] [Config      ] Older configuration version detected (0 instead of 24)
10-19 15:38:03.094  5693  5762 I python  : [WARNING] [Config      ] Upgrading configuration in progress.
10-19 15:38:03.102  5693  5762 I python  : [INFO   ] [Logger      ] Record log in /data/user/0/io.unsigned.sideband/files/app/.kivy/logs/kivy_22-10-19_0.txt
10-19 15:38:03.103  5693  5762 I python  : [INFO   ] [Kivy        ] v2.1.0
10-19 15:38:03.104  5693  5762 I python  : [INFO   ] [Kivy        ] Installed at "/data/user/0/io.unsigned.sideband/files/app/_python_bundle/site-packages/kivy/__init__.pyc"
10-19 15:38:03.105  5693  5762 I python  : [INFO   ] [Python      ] v3.9.5 (default, Oct 19 2022, 13:29:05)
10-19 15:38:03.105  5693  5762 I python  : [Clang 12.0.8 (https://android.googlesource.com/toolchain/llvm-project c935d99d
10-19 15:38:03.106  5693  5762 I python  : [INFO   ] [Python      ] Interpreter at ""
10-19 15:38:03.107  5693  5762 I python  : [INFO   ] [Logger      ] Purge log fired. Processing...
10-19 15:38:03.108  5693  5762 I python  : [INFO   ] [Logger      ] Purge finished!
10-19 15:38:03.113  5693  5762 I python  : [INFO   ] [KivyMD      ] 1.1.0.dev0 (installed at "/data/data/io.unsigned.sideband/files/app/./kivymd/__init__.pyc")
10-19 15:38:03.124  5693  5762 I python  : [INFO   ] [Factory     ] 189 symbols loaded
[... cut ...]
10-19 15:38:26.970  5988  6004 I sidebandservice:  jnius.jnius.JavaException: JVM exception occurred: Didn't find class "android.app.NotificationChannel" on path: DexPathList[[zip file "/data/app/io.unsigned.sideband-1/base.apk"],nativeLibraryDirectories=[/data/app/io.unsigned.sideband-1/lib/arm, /data/app/io.unsigned.sideband-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib, /system/vendor/lib, /product/lib]] java.lang.ClassNotFoundException
10-19 15:38:26.971  5988  6004 I python  : Python for android ended.
```

closed 2022-10-19 by Cinorid

markqvist, 2022-10-19:
Awesome. Please test it out and let me know if you find anything else. If it is actually more or less workable all-round, I could integrate the necessary changes into the primary codebase and set up build steps for supplying `armeabi-v7a` binaries with the normal releases.

markqvist, 2022-10-20:
Hi @Cinorid 

Please take a look at this: https://github.com/markqvist/Reticulum/discussions/123

I updated the `armeabi-v7a` build to the latest `0.2.5` version and created a universal APK that is easier to share (works both on 32 and 64 bit).