markqvist/Sideband #11: Sporadic error message "Could not send the message" with propagated delivery

SebastianObi, 2022-10-21:
Hi,

i have a strange issue.

Sometimes i get this error message "Could not send the message" when i try to send a propagated message.
The message is still delivered successfully and the status of the message in the display is also successful.

I think in the "send_message" function happens some error which triggers the exeption.
Unfortunately, I don't know how to debug this further. It should be possible to activate the debug log somewhere and then display it as a file. I use the apk on Android. Then we could se the "RNS.LOG_ERROR" exeption string.

```
    def send_message(self, content, destination_hash, propagation):
        try:
            if content == "":
                raise ValueError("Message content cannot be empty")

            dest_identity = RNS.Identity.recall(destination_hash)
            dest = RNS.Destination(dest_identity, RNS.Destination.OUT, RNS.Destination.SINGLE, "lxmf", "delivery")
            source = self.lxmf_destination
            
            if propagation:
                desired_method = LXMF.LXMessage.PROPAGATED
            else:
                desired_method = LXMF.LXMessage.DIRECT

            lxm = LXMF.LXMessage(dest, source, content, title="", desired_method=desired_method)
            lxm.register_delivery_callback(self.message_notification)
            lxm.register_failed_callback(self.message_notification)

            if self.message_router.get_outbound_propagation_node() != None:
                lxm.try_propagation_on_fail = True

            self.message_router.handle_outbound(lxm)
            self.lxm_ingest(lxm, originator=True)

            return True

        except Exception as e:
            RNS.log("Error while sending message: "+str(e), RNS.LOG_ERROR)
            return False
```

I used it on my Samsung S20FE with Android 12 and Sideband 0.2.5 beta

markqvist, 2022-10-22:
I have also experienced this very sporadically, but I initially put it down to the "Send" button accidentially being pressed two times, which would cause the second send command to error out. After I implemented disabling the button until the sending completed, I have not experienced it myself, so I wrote it off as completed. But apparently not ;) There might be something else going on here.

I agree that there needs to be a way to view and export the log, and to turn debug logging on and off. It's been on my list for a while, but as you know by now, *priorities*...

Currently, LXMF is on the top of the priority list, especially adding daemon utilities for running propagation nodes and other tasks as services directly, without needing nomadnet, and of course the big one, *actual API documentation*. When that is taken care of, I will return to this :)

In the meantime, if you gather any more details about it, please let me know. I can also supply you with a build with debug logging enabled, then you can at least offload the logs with `adb logcat | grep "python\|sidebandservice"`.

SebastianObi, 2022-10-22:
I know it is always a problem with the priorities and available time.
I appreciate very much your engagement in the development of the whole Reticulum universe.
It all takes a lot of time.

In the meantime I found out how to install and use the development environment to compile Sideband myself :) I will write a short manual and send it to you for the documentation. Then it is easier if someone else wants to code on Sideband.

I may have found an error in the makefile. I will open an issue for this.

markqvist, 2022-10-22:
Great! Thanks a lot for all the feedback and input so far :)

closed 2022-10-25 by markqvist