markqvist/Sideband #12: Error in the Makefile "getrns"

SebastianObi, 2022-10-22:
Perhaps I have found an error. Or do something wrong :)

I have downloaded the latest version of Reticulum and Sideband.

In the Makefile there is the command `;rm ./RNS/Utilities/RNS` at the `getrns:` section.

But this folder does not exist in the download of recticulum. Therefore the Makefile aborts here with an error.

When I delete this part, everything works fine.

markqvist, 2022-10-22:
It is removing symlinks that are created by the build scripts for Reticulum itself, since those symlinks should not be included in the built APKs or wheels, to avoid link recursions. If they are not there on your system, it is because you have not built Reticulum before building Sideband.

So no, it is not a bug, but thanks for mentioning it, in case it was.

markqvist, 2022-10-22:
So just to expand, instead of deleting the command from the makefile, you should just compile (and thereby also run the tests for) the Reticulum release files, before running `make` in the Sideband source tree. Just build Reticulum with `make release` and all tests and everything is run on your system. This is definitely a good idea before compiling Sideband, since it will package the Reticulum distribution from your system directly into the built APK ;)

closed 2022-10-22 by markqvist