markqvist/Sideband #16: Feature request: make rnodeconf available in android

LaneaLucy, 2022-11-30:
Now rnodeconf is integrated into rns, so it should be in the android app. If you now make it available from the GUI, you can update rnode's or init new one's. This would make it very easy to manage everything without need for a second device in a mobile scenario.
Thx for your work

markqvist, 2022-11-30:
Yes, this would be very nice. There's probably a few minor additions that are needed to make it work under the different OS structure of Android, but all in all, it should be totally doable.

The biggest hurdle will probably be getting the esp32 flashing tool to work under Android, but even that is not something I think will be too much of an issue.

So yeah, it should definitely be possible, and I agree, would be very useful.

LaneaLucy, 2022-11-30:
Shouldn't the flashing tool be base on c? If I remember right, you can include c code into android apps

markqvist, 2022-11-30:
For some platforms it is, like AVR and ARM, but for ESP32 it is written in Python.

LaneaLucy, 2022-11-30:
Shouldn't make it this then even easier? I mean at max someone need to replace the serial function in it

LaneaLucy, 2022-11-30:
Found a c implementation that can easy being ported

https://github.com/espressif/esp-serial-flasher

LaneaLucy, 2022-11-30:
And here is a java port

https://github.com/bdureau/ESPLoader

LaneaLucy, 2022-11-30:
If I didn't hate java that much I would try to implement it myself and gave it to you, but I really, REALLY hate java

markqvist, 2022-12-02:
> If I didn't hate java that much I would try to implement it myself and gave it to you, but I really, REALLY hate java

I feel your pain there... I am not a particular fan either.