markqvist/Sideband #17: i2pd not functioning correctly in Sideband

karl-baker, 2023-01-18:
Hi Mark,

I've been informed of something not functioning as expected when trying to use the i2pd daemon within Sideband.

I'm not aware of all details, but wanted to reach out to you and make you aware of this as well. If you could look into this that would be great. If you need more information let me know and I'll see what I can find out.

Thanks.

markqvist, 2023-01-18:
Hi Karl

Thanks for letting me know. It'd be great if you could collect some more details, sinice I don't really have much to go on otherwise, and in my own testing it is working fine.

You do know that you have to run an I2P daemon on the device before launching Sideband, right? So, just to be completely verbose, Sideband does **not** run it's *own* instance of I2P, it uses the I2P instance already running on the system.

The one I've had the most success, and least problems with, is the `i2pd` implementation from here: https://i2pd.website/

It seems to perform a bit better than the Java implementation, at least on Android.

Would you mind making sure that whoever reported this to you is aware of those details?

karl-baker, 2023-01-18:
Hi Mark,

Let me see what I can find out and I'll get back to you.

karl-baker, 2023-01-21:
Hi Mark,

We have ran some tests with the latest version of sideband, and have came to the conclusion that it is working as expected.

Sorry for raising any flags, but now we know that it works.

Thanks for your help.

closed 2023-01-21 by karl-baker

markqvist, 2023-01-21:
No worries, better safe than sorry! Thanks for reporting back!