markqvist/Sideband #18: Quite slow (10-15) seconds loading time.

CutePotatoDev, 2023-01-23:
Application tooks 10 to 15 seconds to load up.
Shut down took shorter time but also not to much satisfying ammount.

markqvist, 2023-01-23:
Can you give some info on what devices these are, and what Android version they are running on?

The first run of the app will usually be quite slow on older devices, since it needs to unpack the various resources and libraries for the first time. It should be noticably faster next time.

CutePotatoDev, 2023-01-23:
Device 1 (Phone):
Samsung Galaxy A50 (SM-A505FN/DS)
Android 11 (One UI 3.1)

Device 2 (Tablet):
Huawei MediaPad M5 Lite (BAH2-W19)
Android 8.0.0 (EMUI 8.0.0)
This one loads slower than first one.

markqvist, 2023-02-09:
Thanks. There does seem to be some issues on devices with older/slower chips. Unfortunately it mostly comes down to the fact that Sideband has to unpack some stuff (the python dependencies and related libraries) into the apps private chache folder on first launch, or whenever the OS clears the app cache. On not-so-fast devices, this takes a while.

Things can *probably* be optimised a bit, and I will definitely have a look at it, but it is not the highest priority right now.

closed 2023-05-04 by markqvist