markqvist/Sideband #20: pip install sbapp on fedora not install

srem1, 2023-02-08:
Building wheels for collected packages: kivy
  Building wheel for kivy (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for kivy (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [723 lines of output]
      [INFO   ] [Logger      ] Record log in /home/ooo/.kivy/logs/kivy_23-02-08_5.txt
      [INFO   ] [Kivy        ] v2.1.0
      [INFO   ] [Kivy        ] Installed at "/tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/__init__.py"
      [INFO   ] [Python      ] v3.11.1 (main, Jan  6 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)]
      [INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
      [INFO   ] [Logger      ] Purge log fired. Processing...
      [INFO   ] [Logger      ] Purge finished!
       <string>:370: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
       <string>:373: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
       <string>:377: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
       warning: kivy/graphics/common.pxi:9:4: 'const_char_ptr' redeclared
       warning: kivy/graphics/common.pxi:23:4: 'size_t' redeclared
      /tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/weakproxy.c: In function ‘__Pyx_AddTraceback’:
      /tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/weakproxy.c:438:62: error: invalid use of incomplete typedef ‘PyFrameObject’ {aka ‘struct _frame’}
        438 |   #define __Pyx_PyFrame_SetLineNumber(frame, lineno)  (frame)->f_lineno = (lineno)
            |                                                              ^~
      /tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/weakproxy.c:12479:5: note: in expansion of macro ‘__Pyx_PyFrame_SetLineNumber’
      12479 |     __Pyx_PyFrame_SetLineNumber(py_frame, py_line);
            |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      /tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/_event.c: In function ‘__Pyx_PyBytes_Equals’:
      /tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/_event.c:19716:13: warning: ‘ob_shash’ is deprecated [-Wdeprecated-declarations]

------
[INFO   ] Kivy: v2.1.0
[INFO   ] Kivy: Installed at "/tmp/pip-install-vj7x_3t8/kivy_517a5bba2bf14100bc7a5e4ffadec816/kivy/__init__.py"
[INFO   ] Python: v3.11.1 (main, Jan  6 2023, 00:00:00) [GCC 12.2.1 20221121 (Red Hat 12.2.1-4)]
[INFO   ] Python: Interpreter at "/usr/bin/python3"
[INFO   ] Logger: Purge log fired. Processing...
[INFO   ] Logger: Purge finished!
[WARNING] stderr: <string>:370: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
[WARNING] stderr: <string>:373: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
[WARNING] stderr: <string>:377: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
[WARNING] stderr: warning: kivy/graphics/common.pxi:9:4: 'const_char_ptr' redeclared 
[WARNING] stderr: warning: kivy/graphics/common.pxi:23:4: 'size_t' redeclared 
[WARNING] stderr: error: command '/usr/bin/gcc' failed with exit code 1
----
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/12/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-12.2.1-20221121/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-offload-defaulted --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.1 20221121 (Red Hat 12.2.1-4) (GCC)

markqvist, 2023-02-09:
Thanks for reporting this. I'll probably have to try and spin up a Fedora VM to check this out. Most likely just a dependency problem.

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-04-27:
Installed kivy development version from git, it kivy was the bigges issue beside missing deps on my Ubuntu.

markqvist, 2023-05-04:
Thanks for reporting back. I am seeing the same problem here with Fedora38. It is not possible to install ``kivy`` via pip, and the only solution currently is to install the dev version from their git repo, until an updated version is released.

The Ubuntu 23.04 issue is different, and stems from the fact that Debian & Ubuntu disallowed using ``pip`` to install packages in their latest versions, forcing people to use the ``pipx`` tool instead, which puts every package in an isolated environment.

While you should be able to install Sideband through ``pipx``, unfortunately the ``pipx`` tool is currently not able to install ``kivy``. Bug report filed here: https://github.com/pypa/pipx/issues/975.

markqvist, 2023-05-04:
Closing this now, as there is unfortunately not anything else I can do to alleviate the problem, and the issue will have to be fixed in ``kivy`` and ``pipx``.