markqvist/Sideband #23: "please wait substantiating reticulum", stuck on launch page

pizzadoglover911, 2023-02-19:
I hope I am correct in starting a new thread. I read over the other open issues but could not acertain a resolution. Upon launch, sideband will not proceed past the launch/landing, as quoted in the title. 

I have left it installed and cleared app data, no change. Uninstalled and re installed, allowed unrestricted battery use/optimization. 

Device is a pixel 3 xl, OEM android build version 12. I did successfully install and open the software on an inexpensive "U318AA" running android 10. So this issue seems specific to my device, whether that be the Pixel model, or my personal device. 

Let me know if there is any specific logs I can pull to help! I am a novice, but this project seems very important, and I can't wait to get into it myself!

Thank you

markqvist, 2023-02-22:
Thanks for reporting this. If you have access to pull an `adb logcat` dump from the device while the program is starting, it would be very helpful!

pizzadoglover911, 2023-02-27:
below is the log pulled while the app was launching, I narrowed down using the term "sideband", hopefully that doesnt remove any important data. I can run again if needed.

<details>
  <summary>logcat pull from android studio while launching</summary>
  
  ### Log
--------- beginning of main
--------- beginning of system
2023-02-26 20:31:56.706  1882-3603  ActivityTaskManager     system_server                        I  START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=io.unsigned.sideband/org.kivy.android.PythonActivity bnds=[1152,1816][1440,2228]} from uid 10389
2023-02-26 20:31:56.731  1882-2015  ActivityManager         system_server                        I  Start proc 19771:io.unsigned.sideband/u0a388 for pre-top-activity {io.unsigned.sideband/org.kivy.android.PythonActivity}
2023-02-26 20:31:56.757 19771-19771 nativeloader            pid-19771                            D  Configuring classloader-namespace for other apk /data/app/~~uB2wweVqOjYJbWK6oyrW0w==/io.unsigned.sideband-UE-SrHF_9msJD6gql0vemQ==/base.apk. target_sdk_version=30, uses_libraries=, library_path=/data/app/~~uB2wweVqOjYJbWK6oyrW0w==/io.unsigned.sideband-UE-SrHF_9msJD6gql0vemQ==/lib/arm64:/data/app/~~uB2wweVqOjYJbWK6oyrW0w==/io.unsigned.sideband-UE-SrHF_9msJD6gql0vemQ==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/io.unsigned.sideband
2023-02-26 20:31:56.765 19771-19771 GraphicsEnvironment     pid-19771                            V  ANGLE Developer option for 'io.unsigned.sideband' set to: 'default'
2023-02-26 20:31:56.791 19771-19791 pythonutil              pid-19771                            V  Unpacking /data/app/~~uB2wweVqOjYJbWK6oyrW0w==/io.unsigned.sideband-UE-SrHF_9msJD6gql0vemQ==/lib/arm64/libpybundle app
2023-02-26 20:31:56.862 19771-19793 SDL                     pid-19771                            V  Running main function SDL_main from library /data/app/~~uB2wweVqOjYJbWK6oyrW0w==/io.unsigned.sideband-UE-SrHF_9msJD6gql0vemQ==/lib/arm64/libmain.so
2023-02-26 20:31:56.863 19771-19793 python                  pid-19771                            I  /data/user/0/io.unsigned.sideband/files/app
2023-02-26 20:31:56.865 19771-19793 python                  pid-19771                            I  /data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip:/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules
2023-02-26 20:31:56.897  1882-1998  ActivityTaskManager     system_server                        I  Displayed io.unsigned.sideband/org.kivy.android.PythonActivity: +187ms
2023-02-26 20:31:56.892 19771-19771 SDLThread               pid-19771                            W  type=1400 audit(0.0:1070): avc: granted { execute } for path="/data/data/io.unsigned.sideband/files/app/_python_bundle/modules/zlib.cpython-39.so" dev="dm-8" ino=394228 scontext=u:r:untrusted_app:s0:c132,c257,c512,c768 tcontext=u:object_r:app_data_file:s0:c132,c257,c512,c768 tclass=file app=io.unsigned.sideband
2023-02-26 20:31:56.904  1882-3603  InputManager-JNI        system_server                        W  Input channel object '95de22b Splash Screen io.unsigned.sideband (client)' was disposed without first being removed with the input manager!
2023-02-26 20:31:56.914  3464-3464  GoogleInpu...hodService com...gle.android.inputmethod.latin  I  GoogleInputMethodService.onStartInput():1930 onStartInput(EditorInfo{inputType=0x0(NULL) imeOptions=0x0 privateImeOptions=null actionName=UNSPECIFIED actionLabel=null actionId=0 initialSelStart=-1 initialSelEnd=-1 initialCapsMode=0x0 hintText=null label=null packageName=io.unsigned.sideband fieldId=-1 fieldName=null extras=null}, false)
2023-02-26 20:31:56.925 19771-19793 python                  pid-19771                            I  Android path ['.', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/site-packages']
2023-02-26 20:31:56.925 19771-19793 python                  pid-19771                            I  os.environ is environ({'PATH': '/product/bin:/apex/com.android.runtime/bin:/apex/com.android.art/bin:/system_ext/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'ANDROID_ART_ROOT': '/apex/com.android.art', 'ANDROID_I18N_ROOT': '/apex/com.android.i18n', 'ANDROID_TZDATA_ROOT': '/apex/com.android.tzdata', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'DOWNLOAD_CACHE': '/data/cache', 'BOOTCLASSPATH': '/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/apex/com.android.i18n/javalib/core-icu4j.jar:/apex/com.android.appsearch/javalib/framework-appsearch.jar:/apex/com.android.conscrypt/javalib/conscrypt.jar:/apex/com.android.ipsec/javalib/android.net.ipsec.ike.jar:/apex/com.android.media/javalib/updatable-media.jar:/apex/com.android.mediaprovider/javalib/framework-mediaprovider.jar:/apex/com.android.os.statsd/javalib/framework [... cut ...]
2023-02-26 20:32:05.419  1882-3603  ActivityManager         system_server                        I  Process io.unsigned.sideband:service_sidebandservice (pid 19806) has died: prcp FGS 
2023-02-26 20:33:55.419  3464-3464  GoogleInpu...hodService com...gle.android.inputmethod.latin  I  GoogleInputMethodService.onStartInput():1930 onStartInput(EditorInfo{inputType=0x0(NULL) imeOptions=0x0 privateImeOptions=null actionName=UNSPECIFIED actionLabel=null actionId=0 initialSelStart=-1 initialSelEnd=-1 initialCapsMode=0x0 hintText=null label=null packageName=io.unsigned.sideband fieldId=-1 fieldName=null extras=null}, false)

 
</details>

markqvist, 2023-03-02:
Ok, so it is the background service that is dying. There's probably some permission violation or similar. Thanks for providing the info, this should make it easier to find a solution!

pizzadoglover911, 2023-03-06:
I assume you aren't talking about the permissions that can be adjusted in the user settings of Android, though I allowed all I could with no change.

In my phone graveyard I managed to get Sideband running on a pixel 2 xl, running android 11. No issues there.

I'm starting to think it is some app I've got installed or a specific setting on my phone.

markqvist, 2023-03-06:
Thanks for the feedback. It's a bit weird, since I know it has previously been working on Pixel phones with Android 12. It could be that something has changed though, that broke it. If I provide you with a debug build, with full debug output logging, would you be able to do an `adb logcat` from that?

pizzadoglover911, 2023-03-06:
Yes I could get that done. Is narrowing down the log by search term 'sideband' effective? My concern was if that log showed any identifying info about the phone. I am equally concerned that the lack of info could cause the log to be less useful to you.

I have functionality on other devices at the moment which is awesome. I am still a little unsure of how sideband is connecting to my Rnodes, and if that connection is effectively transmitting data, but that is outside the scope of this issue and I may find the answer as I dig further through your documentation.

markqvist, 2023-03-06:
Here's the debug APK. Just for good measure, uninstall the old one before installing this. Just to make sure there's not any messed up configs or caches from the old one.

https://unsigned.io/dl/sideband-0.5.2-arm64-v8a_armeabi-v7a-debug.apk

pizzadoglover911, 2023-03-07:
The two outputs are attached. I ran once without grep 'python' and once with. Both logs were performed on a fresh install of your above package. 

[adb_logcat_grep_sideband.txt](https://github.com/markqvist/Sideband/files/10904180/adb_logcat_grep_sideband.txt)
[adb_logcat_grep_python_sidebandservice.txt](https://github.com/markqvist/Sideband/files/10904181/adb_logcat_grep_python_sidebandservice.txt)

Let me know how you make out!

markqvist, 2024-01-03:
Assuming this is fixed by the earlier restructuring to the permissions and Android API calls.

closed 2024-01-03 by markqvist