markqvist/Sideband #24: Sideband as interface for NomadNet in Android (Termux)

Dant3s, 2023-02-26:
Hi! :wave: 

I notice that when starting nomadnet from termux it can't run since it can't talk to the native part of the android/usb/bluetooth, but launching nomadnet after starting Sideband does the trick. 

This would mean that there are two profiles running. Sideband Client and NomadNet client. Even if the ids are different. 

Is there any way to run sideband just to use as a interface for nomadned? without using it as a client?

markqvist, 2023-02-27:
What you have noticed is correct :) It is possible to use the RNS instance launched from Sideband, which does have access to USB and Bluetooth, for other Reticulum programs inside Termux.

Currently, it is not possible to just run Sideband as system-wide RNS instance, but the even better solution, which is to have a standalone Android RNS Daemon app, is already on the roadmap.

Dant3s, 2023-02-27:
Sweet. Thanks! for the info!

closed 2023-02-27 by Dant3s