markqvist/Sideband #25: Dependency problem with P4A

faragher, 2023-04-03:
Trying to compile Sideband has been problematic, mostly because this is way above my Python paygrade. I have found and solved one major issue that I don't believe is my fault.

The current system attempts to build referencing NPK 23b, while Kivy pushed the minimum version to 25. I tried faking it, but that failed. I finally found the incompatibility and tracked down the configuration in ./sbapp/venv/lib/python3.9/site-packages/buildozer/targets/android.py (This is a runtime file, and I have no idea where it pulls in the data)

By defining the branch name as "release-2022.09.04" the previous release is pulled and the system functions as intended. 

I have absolutely no idea how to definitively fix this, but with this information it's likely to be a straightforward fix to someone familiar with buildozer etc.

As a warning, it's still not compiling entirely, but I believe this shouldn't have knock-on effects, and simply restores functionality to intended levels.

faragher, 2023-04-05:
Closing as the entire build process appears broken. This may be because of downloading rolling releases that may have depreciated entire SDK versions or simply due to downloading dependencies on the fly, but after four days of tracking down error messages, I have reached the least helpful gradle error messages possible, and am giving up on compiling the project altogether.

closed 2023-04-05 by faragher

markqvist, 2023-04-05:
The whole buildozer/p4a build system went bonkers after the last few updates, with incompatibilities between everything leading to all kinds of insane situations. I had to seriously patch up things manually to get it all working in my end. Crossing fingers for a fixed release from them soon.

faragher, 2023-04-05:
Thanks, it meshes with my experience. Kinda trying to get enough of a C# library to make things work for testing, but the system is at the point where I need a mobile app for testing, and, well, Reticulum needs Reticulum. I suppose I could set up a laptop or a Pi for mobile testing, but that's a hack for testing.

markqvist, 2023-04-06:
I'm working on an easier way to include Reticulum in Android projects, so the whole buildozer/p4a system wont be necessary, but I honestly don't know when that will be ready.

faragher, 2023-04-06:
I remember loading IPX TSRs in DOS. I was a little young to really understand the network wars and really appreciate what a game changer routed networking was, especially when it was distributed with an OS. I've watched this project for years, and I trust your development map. 

In the meantime, if I make something useful, I'll share it.

markqvist, 2023-04-07:
Thanks, that is definitely appreciated! Any work done towards that end is really valuable right now.

It *is* possible to include pre-compiled shared libraries in android projects, and the goal would be to have a loadable shared library/executable, that contains RNS, which native android apps can then easily interface with.

faragher, 2023-04-07:
Having a library that interfaced with a daemon would be ideal for mobile for the sake of multiple apps. But I know basically nothing about Android development. Being able to interface with the RNode via serial commands, especially over bluetooth, makes it just a library away from working on almost anything, but it's still that library. Going from Python to C# has basically had me walking through the code trying to understand the types runtime defined variables that Python abstracts away. The API document has been a big help, but there's a lot of work defining classes before I can see if I'm on the right path.