markqvist/Sideband #28: Revisit java patches in patches directory

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-08-31:
Hi,
looked briefly on HIDDeviceUSB  inside Reticulum and seems like some fixes have been already merged into upstream. 

Also upstream now looks a bit different then files in patches, over time there have been more contributions most likely.

So there are tasks:
1, Remove patches if upstreamed and try SideBand on Android without HIDDeviceUSB and others found.
2, Build apk
3,Ensure it works
4, If some patches are still needed, try to upstream them.

Did i miss something?

markqvist, 2023-08-31:
Thanks for the suggestions, that is definitely worth having a look at. You're very welcome to help out with this if you want to!

Getting buildozer and python-for-android up and running can sometimes be a little finicky due to various buildozer and Android NDK versions and build dependencies on your system. I'd recommend reading and following these resources:

- https://github.com/kivy/buildozer
- https://github.com/kivy/kivy/wiki/Android-SDK-NDK-Information

Once buildozer is set up correctly, you should simply be able to build Sideband with the included makefiles.

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-09-01:
Having trouble recognizing this issue. May/might be related to android NDK/SDK version combination.  But no idea what to do with it.   

Any idea how to deal with this error? Thank you

In file included from /home/petr/.buildozer/android/platform/android-ndk-r25b/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/limits.h:38:
/usr/include/x86_64-linux-gnu/sys/cdefs.h:64:6: error: function-like macro '__GNUC_PREREQ' is not defined
# if __GNUC_PREREQ (4, 6) && !defined _LIBC
     ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:78:10: error: function-like macro '__GNUC_PREREQ' is not defined
     && (__GNUC_PREREQ (3, 4) || __glibc_has_attribute (__nothrow__))
         ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:84:31: error: function-like macro '__GNUC_PREREQ' is not defined
#  if defined __cplusplus && (__GNUC_PREREQ (2,8) || __clang_major >= 4)
                              ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:146:34: error: function-like macro '__glibc_clang_prereq' is not defined
#if __USE_FORTIFY_LEVEL == 3 && (__glibc_clang_prereq (9, 0)                  \
                                 ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:204:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (4,3)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:282:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,96) || __glibc_has_attribute (__malloc__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:290:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (4, 3)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:299:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (4, 9) || __glibc_has_attribute (__alloc_align__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:309:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,96) || __glibc_has_attribute (__pure__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:316:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,5) || __glibc_has_attribute (__const__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:322:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,7) || __glibc_has_attribute (__unused__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:331:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (3,1) || __glibc_has_attribute (__used__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:340:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (3,2) || __glibc_has_attribute (__deprecated__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:349:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (4,5) \
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:363:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,8) || __glibc_has_attribute (__format_arg__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:373:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (2,97) || __glibc_has_attribute (__format__)
    ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:385:6: error: function-like macro '__GNUC_PREREQ' is not defined
# if __GNUC_PREREQ (3,3) || __glibc_has_attribute (__nonnull__)
     ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:398:6: error: function-like macro '__GNUC_PREREQ' is not defined
# if __GNUC_PREREQ (4, 9) || __glibc_has_attribute (__returns_nonnull__)
     ^
/usr/include/x86_64-linux-gnu/sys/cdefs.h:407:5: error: function-like macro '__GNUC_PREREQ' is not defined
#if __GNUC_PREREQ (3,4) || __glibc_has_attribute (__warn_unused_result__)
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.

markqvist, 2023-09-02:
Please include pasted terminal output in a code block so it's readable :)

qwertyuiopzxcvbnmlkjhgfdsa1, 2023-09-05:
I am thinking about dockerfile for building artifacts. 

@markqvist  Could you past here output?  

`pip freeze > requirements.txt`

markqvist, 2024-01-03:
The build system needs to have a current version of the RNS and LXMF repositories placed in the same folder as the root Sideband source directory, like so:

```
--+ Reticulum
--+ LXMF
--+ Sideband
  |
  +-- sbapp
  +-- (more files)
```

The build system copies these into the Sideband package for Android. You could probably just clone them from github during the docker setup process, or mount them from the host FS.

markqvist, 2024-01-03:
Since this is not actually causing any issues, is currently working, and is veering more into a thread about dockerising the build, I am going to move it to a discussion.