markqvist/Sideband #29: WSL Sideband

n4ijk, 2023-09-04:
I am using WSL on Windows 11 to run Ubuntu and ran into two issues to get sideband running.  

1) "MTDEV not supported by your version of linux" which I solved with sudo apt-get install libmtdev-dev

2) "Unable to find any valuable Cutbuffer provider" which was solved with sudo apt-get install xclip

Works fine with those packages installed so far. This is my first time participating with a project like this - so if I've done this wrong please let me know!

markqvist, 2023-10-03:
You can run Sideband directly through Python under windows now, by simply installing it with `pip`.

- Download and install the latest version of Python for windows.
- During the installation, be sure to include the 'pip' utility.
- During the installation, be sure to add Python and related utilities to your `PATH` environment variables (it is a checkbox option in the installer)
- Install Sideband via `pip` in the Windows command prompt or Powershell: `pip install sbapp`
- Run sideband from a command line: `sideband`

Remember that the Reticulum `AutoInterface` is not currently supported under Windows, so you will need to manually configure some sort of connectivity by adding one or more interfaces to the Reticulum config file on your system in `C:\Users\USERNAME\.reticulum\config`.

markqvist, 2023-10-03:
And thanks for reporting those workarounds for anyone else who wants to get it running under WSL!

Would you mind sharing if the AutoInterface works with that configuration?

closed 2023-10-03 by markqvist

n4ijk, 2023-10-03:
My pleasure! The entire sideband app worked just fine with the work arounds. I didn’t play with it for too long but wanted to give it a try.