markqvist/Sideband #31: Change the license

GrimPixel, 2023-09-23:
Creative Commons licenses are not for programs.
https://opensource.stackexchange.com/questions/1717/why-is-cc-by-sa-discouraged-for-code

markqvist, 2023-09-23:
I am perfectly well aware of such considerations. If you think I simply smacked on a random license on my software without any consideration or thought, I can reassure you that is not the case.

A shallow statement like "X is not for Y", and a link to a SE questions is not really useful, and all I can do is close this "issue".

If you want to discuss or propose something, make your own statements and arguments on why you think something should be changed, and how you think it would improve things.

Decrees by proxy of a hyperlink is not how we do things here.

closed 2023-09-23 by markqvist

EternityForest, 2024-07-01:
The original issue seems to be fairly useless and shallow, but I think the point does stand on its own, enough to be worth discussion.

Is this explicitly intended to not be usable for businesses of any size to communicate, and to not ever be combined with GPL licensed software?

markqvist, 2024-07-15:
### First, to clear up any confusion:

- *Reticulum*, the protocol, is in the public domain.
- The Python reference implementation, *Reticulum Network Stack* is MIT licensed.
- The LXMF protocol is MIT licensed, as is the reference implementation I have released.
- Other LXMF clients than Sideband exist under GPL and MIT license.
- The *Sideband* application specifically is currently licensed CC-BY-NC-SA-V4, with an additional clause on the compiled binaries that I provide.

From the [License](https://github.com/markqvist/Sideband?tab=readme-ov-file#license) section of the repo / Sideband page:

> Permission is hereby granted to use Sideband in binary form, for any and all purposes, and to freely distribute binary copies of the program, so long as no payment or compensation is charged or received for such distribution or use.

Sideband is *one* client application built on Reticulum and LXMF, that I wrote with pretty specific goals and usecases in mind. It's freely usable by *every person on the entire planet*, and the licensing terms specifically seeks to remove any barriers to access, by disallowing anything *but* free **and** gratis redistribution.

I really cannot see how there are any obstacles to adoption in this context.

### Secondly, to elaborate on more specific situations:

Can you use Sideband within an organisation or business? Sure, you just have to use the compiled binaries that I provide, which will automatically make any users aware that they have access to the source code and further updates to the system, and most importantly this: That *they* - the actual users - own the system and application, not the corporation / business / organisation that may want to appear as if they are "providing" it to them. They are also made aware of the technology stack that makes this possible, and where it came from, so that they understand they have agency and co-ownership.

As such, one of the few things that is **not** permitted within these licensing terms are for some business to download the source code, rebrand the program, and resell it as their own innovation, either internally or externally. Doing that would fall within "commercial use", which the CC license forbids.

Therefore, if you want to do something like that, you a very welcome to negotiate licensing terms with me on the specific basis of whatever it is you want to do :) Or just write your own LXMF client from scratch that suits your needs.

I'm certain that most people here can appreciate why I set up the licensing this way.

msglm, 2024-08-28:
I would have liked to package this program for [GNU Guix](https://guix.gnu.org/), but the [freedom restricting license](https://www.gnu.org/philosophy/free-sw.html) prevents this. Please reconsider the license so more users may make use of and use your application.

EternityForest, 2024-10-16:
I'm not a lawyer, but ss I understand it even the regular GPL preserves freedom if resold. 

You can sell a copy, but you must provide source code of all your changes, and the modified version becomes GPL licensed too, you can't distribute a nonfree rebrand. The GPL was very specifically meant to prevent anyone from trying to take away anyone's freedom, while allowing business models like preinstslled hardware, consulting, and support.

serialrf433, 2025-12-17:
I just found this closed issue. 

I had opened a discussion here that got directly closed: https://github.com/markqvist/Reticulum/discussions/1041
I answered more in detail here: https://github.com/markqvist/Reticulum/discussions/790#discussioncomment-15274280

I see in this issue here the first argument that have been written down that should be fixed by the NC license. https://github.com/markqvist/Sideband/issues/31#issuecomment-2229553025

Great to have finally something to work with.

> one of the few things that is not permitted within these licensing terms are for some business to download the source code, rebrand the program, and resell it as their own innovation, either internally or externally.

Do i understand your idea, by trying to prevent that, you would like to reach the goal you mentioned before?

 > It's [Sideband] freely usable by every person on the entire planet, and the licensing terms specifically seeks to remove any barriers to access, by disallowing anything but free and gratis redistribution.

Since the license change in 2022 there is now a fork with a rebranding. Its named openCom-Companion. https://github.com/liberatedsystems/openCom-Companion
The license is unchanged in openCom-Companion. It would highly have been unchanged if it were for example AGPLv3.

Lets imagine there would be now openCom-Companion as a fork, AGPLv3 licensed where jacobeva is selling the precompiled binary. I think you would not like that because then openCom-Companion would not be free and **gratis**. And if something cost money, it would not be freely available to every person on the entire planet, right? Did i understand your point here with this example?
Why do you even care about openCom-Companion in such a hypothetical situation? Just ignore it. Everyone can still download and build openCom-Companion freely and gratis on their own. Everyone can fork openCom-Companion, rename it and release this fork. But likely no one would do that. There is still the original named Sideband.

I point here to the endless discussion at [Signal](https://github.com/signalapp/Signal-Android)<->[Molly](https://github.com/mollyim/mollyim-android) fork. Also point to the endless discussion why [Signal] is not available on F-Droid.

But Sideband is not just an Android app. Its also not available in the general packages Linux distributions use to provide software. This is really blocking the masses.
Part of your words **freely usable by every person on the entire planet** should be also simplicity to install and use it. That is why, as much as i dislike it, there should be iOS version in the Apple store. There should be a Google Play Store version. It should be available in ubuntu packages, on opensuse, on fedora, on debian, on F-Droid, ... just everywhere installable with the build in GUI software installer. Yes, also even the Microsoft Windows store should have Sideband in it. But Sideband is currently not available on all those places. Developers from one of the most free Linux distributions, GNU Guix, even joined here and also mention the issue to not be able to provide it in their FLOSS OS because of the license: https://github.com/markqvist/Sideband/issues/31#issuecomment-2315917558

Sideband should really just be AGPLv3 licensed to solve that.

serialrf433, 2025-12-19:
I have not attacked anyone. The LLM you used is making things up to blame me having done something i have not done. Combined with overall no answers to the points i wrote about.

> The openCom-Companion Lies
> Your Complete Fabrication

I begun my sentence in my message with this words: **Lets imagine there would be**
* Of course some imaginary situation is from the fact point of view a lie
* Of course its a complete fabrication

That is the point of explaining something based on imagination. Based on things that are not true. Based on fantasy.

I have spend time to try to find as much information about why Sideband have this license and tried to point to the current issues it have. One of the very few information sources about the Sideband license are here. I have nothing to do with the person who opened this topic here. And i have nothing to do with closing the issue some time ago. I also have nothing to do with the date and time all that have happened. I search for information, read, and when something is not clear, i ask.

The LLM allowed me to create a project named Sideband with AGPLv3 license using the same protocol 😮‍💨

I do not know what the goal of not answering the things i wrote should have. It have no benefit to the project and its users.

> Delete and ban - this type of nonsense doesn't deserve engagement

Bringing Sideband with a different license to app-stores (Apple, Google Android, MacOS, Microsoft, F-Droid, Debian repo, ...) would bring benefit to this goal: **freely usable by every person on the entire planet**

When i try to point out the things that are unclear as detailed as possible and spend time searching for the answers on my own as good as i can before asking questions and when i then ask questions i get such a answer, then i do not have the feeling i could benefit somehow at this license topic.