markqvist/Sideband #32: Add to official F-Droid Repo / Website

hhglae, 2023-10-03:
Currently this Android App is not in F-Droid. It seems to be free software and i cant find a technical reason why its not in F-Droid. Could you add it to official f-droid so it get public build logs from public servers?
Having to download a apk somewhere from the internet or add custom repo like https://android.izzysoft.de/repo/info described here https://unsigned.io/website/sideband/ is not really recommended way for installing software.

The Idea behind https://reproducible-builds.org/ is good and important and F-Droid makes a simple to use solution for such things.

markqvist, 2023-10-03:
Thanks for bringing this up. I understand that it would be convenient for some users to install Sideband via F-Droid.

Downloading an APK and installing from the official github repo is completely fine.  That it is "not really a recommended way for installing software" is a bit of a strange myth, mostly created by certain conglomerates with near-monopolies on mobile software distribution.

The APK is signed with my signing keys, so once you have installed it from this official repo, you can be sure you didn't get a modified version from somewhere else. On the other hand, if you think I'm going to include malware in the app, it's not really going to help you to download it from an app store ;)

Either way, I agree that it would be great to have it on F-Droid, but there are certain technicalities involved in that. I am not sure that their build system would even support compiling Sideband, since it uses buildozer and python-for-android to compile an included Python interpreter to run both Reticulum and the user interface.

And correct me if I'm wrong here, but doesn't building through F-Droid imply that they will use *their* signing keys for the app? That would probably also have implications for the built-in distribution mechanisms in Sideband (the repository server).

Given the low amount of benefit it creates, this is not something that I have the resources to prioritize supporting at this point.

closed 2023-10-03 by markqvist

marek22k, 2023-11-16:
> And correct me if I'm wrong here, but doesn't building through F-Droid imply that they will use _their_ signing keys for the app? That would probably also have implications for the built-in distribution mechanisms in Sideband (the repository server).

If you use [reproducible builds](https://f-droid.org/docs/Reproducible_Builds/), you can also sign the app with your own key - then F-Droid only checks whether you have built the app from the source code mentioned.

For me, having an app in F-Droid brings trust above all else. F-Droid guarantees that the app is 100% FLOSS and often also removes trackers.

However, the app is also available in F-Droid via IzzyOnDroid's repo (if you use Izzy's repo).

ProgrammerWhoPrograms, 2024-07-26:
Has anyone here heard of [Accrescent](https://accrescent.app/)? It's a beta/alpha app store focussed on security that serves APKs signed by the original devs.

licaon-kter, 2025-12-11:
As the app is not FLOSS (https://spdx.org/licenses/ -> CC-BY-NC-SA-4.0) it can't be included in F-Droid

But, the developers can just host their own F-Droid repository so users could install and get updates