markqvist/Sideband #33: Telemetry stops when application gets backgrounded,

faragher, 2023-11-02:
This is likely unavoidable, but in an overnight test the collector and one sender were locked and left to sit while another sender was used as usual. Likely due to memory issues, the Sideband app was put in the background (and required a second pass through "initializing Reticulum" and did not send telemetry. This occurred during the use period, not during the overnight hours. 

Obviously if the program isn't running there's nothing to be done, but is there a possible solution to keep the program in memory or otherwise persist during periods of downtime?

faragher, 2023-11-02:
I would've never guessed that, but it makes sense. I suggest seeing how Traccar does it, since that continuously updates in the background (and is the cause of my short battery life!) - https://github.com/traccar/traccar-web

Their web and manager repos are basically just browsers for their web service manager AFAIK. 

Arbitrary sensors, particularly through the RNode itself, are really great features. 

This does make me wonder, though, why did the other two systems keep working? Is there a configuration option I've changed? I change so many configurations . . .

markqvist, 2023-11-02:
Good idea! I'll check out their code.

Definitely check battery optimisations in the Android settings. Even though Sideband runs a service in the background, Android can still put it to sleep if you don't disable battery optimisations for Sideband. So it might be something like that. Or, maybe just the memory manager shutting down the UI part of the app to free memory, as you said.

markqvist, 2024-01-03:
Added in version `0.7.7`.

closed 2024-01-03 by markqvist