markqvist/Sideband #34: What ist the background for ICMP to 118.197.115.0:0?

p-f-w, 2023-11-11:
Communication over ICMP?

Only a question because i have seen this address before installation of sideband, but is not possible to identify the app caused the ICMP via UID -100 in log of AFWall+ by myself.

markqvist, 2023-11-11:
There is nothing in Sideband that should be sending ICMP traffic.

>  i have seen this address before installation of sideband

Why do you assume it is coming from Sideband, when you say that you've already seen packets to the address before installing Sideband?

On what basis are you concluding it is coming from Sideband?

p-f-w, 2023-11-12:
Because AFWall+ 3.6.0 is listening it in log for Sideband on start, only when Sideband is starting until Sideband is mean must end because it is "no longer connected". The other ICMPs are further running under UID -100 sometimes. 

Is my first try with Sideband, could be possible, thats were a regulary connection and an other app uses the same way.

Other connections are filtered here by a VPN, but there the app does only something , download from GitHub for local Repository, maps fand pictures from  openstreemap.org and virtualearth.net, until today 10:02, a try to a ipv6 address (port 29716), nothing was looking like a connect to the network.

markqvist, 2023-11-12:
The ICMP traffic is coming from somewhere else on your device. No user-installed application on Android will get UID `100` assigned.

Most likely you are seeing tracking software from the phone vendor calling back to a server in China, where that IP is located.

If you find any *real* indication that Sideband is sending unwarranted traffic to anywhere, please do report it here, since that would be seriously out of intended behaviour for the application.

But this is a false interpretation, closing.

closed 2023-11-12 by markqvist