markqvist/Sideband #35: 0.7.7 not connecting via BT on Android 12 (Redmi Note 10 pro)

FD0815, 2023-12-18:
The same issue occurs on multiple RNodes. (Heltec, Lora32 v1 -- and my friend has several T-beams that appear to have the same issue with both a Pixel 5 XL and a 2 XL, although I haven't looked at his logs yet.)  

> 12-18 00:37:12.243 31150 31168 I sidebandservice: [2023-12-18 00:37:12] [Notice] Configuring Reticulum instance...
> 12-18 00:37:12.243 31150 31168 I sidebandservice: [2023-12-18 00:37:12] [Notice] Not enabling Reticulum Transport
> 12-18 00:37:12.243 31150 31168 I sidebandservice: [2023-12-18 00:37:12] [Notice] Purging stale messages... /data/user/0/io.unsigned.sideband/files//app_storage/sideband.db
> 12-18 00:37:14.180 31150 31168 I sidebandservice: [2023-12-18 00:37:14] [Error] Could not open serial port for interface RNodeInterface[RNodeInterface]
> 12-18 00:37:14.180 31150 31168 I sidebandservice: [2023-12-18 00:37:14] [Error] The contained exception was: Could not connect to any Bluetooth devices
> 12-18 00:37:14.180 31150 31168 I sidebandservice: [2023-12-18 00:37:14] [Error] Reticulum will attempt to bring up this interface periodically

Pairing works. SB on either Linux or Windows can connect both via BT or USBserial; and on an older PH-1 running Android 10 everything works, so I don't think it's the RNodes.

Via BT serial terminal I can connect, this is the logcat output for the MAC address

> 12-18 01:12:09.889 25404 25437 D BluetoothRemoteDevices: aclStateChangeCallback: Adapter State: ON Connected: 30:AE:A4:56:15:22
> 12-18 01:12:09.908  1895  1895 D CompanionDeviceManagerService: onDeviceConnected(address = 30:AE:A4:56:15:22)
> 12-18 01:12:09.909  1895  1895 I CompanionDeviceManagerService: onDeviceNearby(justAppeared, address = 30:AE:A4:56:15:22)
> 12-18 01:12:09.920  3674  3674 I AIVS    : BluetoothBase:onConnection Find device :30:AE:A4:56:15:22 not exist.
> 12-18 01:12:09.928  3674  3674 I AIVS    : BluetoothBase:onReceive: ACTION_ACL_CONNECTED,  device : BluetoothDeviceInfo{ mCurrentOpType: 1 Device Edr ='RNode B47E; 30:AE:A4:56:15:22 vid:0xffff pid:0xffff MandatoryUpgrade:false channelStatus:0bleStatus:0 sppStatus:0 authStage:0 PowerMode:0 mBackUpMode:-1 isDeviceInfoUpdated:false authCmd:false FailedReason:0x0}
> 12-18 01:12:09.929 25404 25404 I BluetoothPhonePolicy: processDeviceConnected, device=30:AE:A4:56:15:22
> 12-18 01:12:09.930 25404 25404 D BluetoothDatabase: setConnection: device=30:AE:A4:56:15:22 and isA2dpDevice=false
> 12-18 01:12:09.930 25404 25404 D BluetoothDatabase: Updating last connected time for device: 30:AE:A4:56:15:22 to 12
> 12-18 01:12:09.930 25404 25404 D BluetoothDatabase: updateDatabase 30:AE:A4:56:15:22
> 12-18 01:12:09.939  5282  5282 I GH.WifiBluetoothRcvr: Connection action: android.bluetooth.device.action.ACL_CONNECTED, device 30:AE:A4:56:15:22
> 12-18 01:12:09.946  5282  6413 I CAR.BTCapsStore: Checking device with address 30:AE:A4:56:15:22
> 12-18 01:12:09.972 25404 25437 D BluetoothRemoteDevices: Skip name update for 30:AE:A4:56:15:22
> 12-18 01:12:09.997  5282  6413 I CAR.BTCapsStore: Checking device with address 30:AE:A4:56:15:22
//disconnect
> 12-18 01:12:19.912  1895  1895 D CompanionDeviceManagerService: TriggerDeviceDisappearedRunnable.run(address = 30:AE:A4:56:15:22)
> 12-18 01:12:20.939 25404 25437 D BluetoothRemoteDevices: Battery level was never set or is already reset, device=30:AE:A4:56:15:22
> 12-18 01:12:20.939 25404 25437 D BluetoothRemoteDevices: aclStateChangeCallback: Adapter State: ON Disconnected: 30:AE:A4:56:15:22
> 12-18 01:12:20.956  1895  1895 D CompanionDeviceManagerService: 30:AE:A4:56:15:22 disconnected w/ reason: (1100) Local request
> 12-18 01:12:20.956  1895  1895 D CompanionDeviceManagerService: onDeviceDisconnected(address = 30:AE:A4:56:15:22)
> 12-18 01:12:20.979 25404 25404 I BluetoothMapService: sRemoteDevice :null device:30:AE:A4:56:15:22 

All permissions for SB are granted and battery optimizations are off. 
Not sure if I missed any step.

FD0815, 2023-12-18:
also this:

> 12-18 01:31:49.846 25404 27283 W ServiceManager: Permission failure: android.permission.BLUETOOTH from uid=10335 pid=11358

Indeed, in the permission list of the installed app android.permission.BLUETOOTH is missing, even though it's declared in the manifest. 
Could it be that on Android 12+ installs this gets snipped but it instead expects the android.permission.BLUETOOTH_SCAN (which is not declared here)?

markqvist, 2023-12-23:
The ˋandroid.permission.BLUETOOTH_SCANˋ permission should not be necessary, since Sideband doesn't actually do any scanning for devices, but only connects to RNodes that have already been discovered and paired via the system Bluetooth settings.

What is actually causing it is currently unclear to me, though. It has been working perfectly fine on Android 12 previously, but something in a recent release *may* have changed that, and I currently don't have an Android 12 device to test with - only 8.1, 9, 11, 13 and 14, which all work fine. The logs from the Pixel phones would definitely help!

Just to get the obvious out of the way, can you check that you actually enabled "Connect via Bluetooth" in Hardware -> RNode settings?

markqvist, 2024-01-03:
According to the [Android SDK docs](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions) is correct. The `android.permission.BLUETOOTH` permission is only needed on Android < 12 (API level 30 and below), and Sideband includes it in the manifest only for API level 30 and below. For API levels above, the other relevant Bluetooth permissions are included.

Since nobody else have chimed in, and I haven't heard anything else indicating this is actually a general bug for Android 12, I am closing this due to lack of feedback for now. Feel free to comment with more information if available.

closed 2024-01-03 by markqvist

markqvist, 2024-05-29:
BLE host connection for RNodes aren't "officially" supported in core RNS or Sideband yet. Both nRF and ESP32-S3 based boards are BLE *only*, and until a BLE-capable interface driver is merged into RNS, those devices are serial/USB only for now.

(except for some cases, for example on Linux, where you can indeed use BLE, and with some of the development versions that already have partially implemented BLE support)

markqvist, 2024-05-29:
If you want to help out with the BLE support, take a look at the work going on over at the [RNode Firmware CE repository](https://github.com/liberatedsystems/RNode_Firmware_CE)!

shodan8192, 2024-05-29:
OK, thanks for clarifications. I'm already looking at new RNode repo :)