markqvist/Sideband #39: Allow adding more than one interface for TCP

jooray, 2024-01-01:
I am trying to use a more reliable node network and I would like to connect to more than one TCP server. The GUI in Sideband (on Android) only allows one TCP connection. If I understand it correctly, it is just a user-interface issue, I can have as many TCP interface configurations as I want.

markqvist, 2024-01-02:
Correct, if there was access to the complete Reticulum configuration file used by the internal RNS daemon of Sideband, you could have any number of interfaces.

What a good way to do this is, without making it super confusing for end users, is open for debate though.

We've previously had discussions about having a standalone RNS daemon app for Android, that would allow more advanced configurations by just editing the config file, like on desktop/servers. Personally, I tend to think of that as a better approach than making the configuration options in Sideband more complex, but I am open to any discussion and ideas here.

jooray, 2024-01-02:
I'm no UX/GUI guru, but I would add a "+" to the bottom saying "Add another TCP interface" and that would duplicate the whole configuration.

markqvist, 2024-01-03:
While I definitely like the idea of having this available as an option, after considering this I will have to place it as out of scope for the current development goals of Sideband.

Realistically, this feature will require a substantial revamp of the interfaces configuration, where each interface can be added as a full-fledged, fully configurable entity in a list, and be configured individually. This is a major feature rewrite, and not something I can prioritise currently.

closed 2024-01-03 by markqvist