markqvist/Sideband #40: Keyboard suggestions don't seem to work on message input

neutralinsomniac, 2024-01-17:
I looked into this a *bit* and this seems possibly related to a change in kivy's [TextInput](https://kivy.org/doc/stable/api-kivy.uix.textinput.html) widget: "Changed in version 2.1.0: [keyboard_suggestions](https://kivy.org/doc/stable/api-kivy.uix.behaviors.html#kivy.uix.behaviors.FocusBehavior.keyboard_suggestions) is now inherited from [FocusBehavior](https://kivy.org/doc/stable/api-kivy.uix.behaviors.html#kivy.uix.behaviors.FocusBehavior)."

I notice that you're using KivyMD, so I looked into how they handle [FocusBehavior](https://kivymd.readthedocs.io/en/0.104.1/behaviors/focus/index.html) but I see no mention of the `keyboard_suggestions` property there, and I'm having a heck of a time getting my attempted patched version of Sideband to run on MacOS to test it (it just pretends to start then immediately exits).

markqvist, 2024-01-17:
Please see [this PR](https://github.com/markqvist/Sideband/pull/38) and [this commit](https://github.com/markqvist/Sideband/commit/d87058517813a175282326e8456eb70c62394e94) for reference.

It's a pretty annoying bug, and the fix is not so trivial it seems, since setting those parameters then causes the bug that was fixed in https://github.com/markqvist/Sideband/commit/d87058517813a175282326e8456eb70c62394e94.

markqvist, 2024-01-17:
If anyone can figure out a method that fixes both, it would be pretty great :)

closed 2024-02-29 by markqvist