markqvist/Sideband #42: Stuck on "Please Wait Substantiating Reticulum" (Class not found)

Doodiepants, 2024-03-19:
**Describe the Bug**
On the first startup, the app is stuck on the "Please Wait Substantiating Reticulum" screen.

**To Reproduce**
Start the app on a clean install.

**Expected Behavior**
I expect the app to start.

**Logs & Screenshots**
Logcat grep sideband
```
03-19 09:10:49.666  1759  4986 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=io.unsigned.sideband/org.kivy.android.PythonActivity bnds=[229,79][442,372]} from uid 10189
03-19 09:10:49.679  5154  5154 I Freeform: onTaskCreated taskId=790 cmp=ComponentInfo{io.unsigned.sideband/org.kivy.android.PythonActivity} target=null
03-19 09:10:49.680  1327  1575 E ANDR-PERF-LM: GamePerfEngine: runAsync() 477: Hint is for io.unsigned.sideband/org.kivy.android.PythonActivity/20240214 = 0x1003 PID=1759
03-19 09:10:49.694  1759  2049 I ActivityManager: Start proc 27825:io.unsigned.sideband/u0a690 for pre-top-activity:com.sonymobile.launcher {io.unsigned.sideband/org.kivy.android.PythonActivity}
03-19 09:10:49.720 27825 27825 D nativeloader: Configuring clns-4 for other apk /data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/base.apk. target_sdk_version=30, uses_libraries=, library_path=/data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/lib/arm64:/data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/io.unsigned.sideband
03-19 09:10:49.725 27825 27825 V GraphicsEnvironment: ANGLE Developer option for 'io.unsigned.sideband' set to: 'default'
03-19 09:10:49.810  1759  2030 I ActivityTaskManager: Displayed io.unsigned.sideband/org.kivy.android.PythonActivity: +140ms
03-19 09:10:50.130 27825 27852 V python  : extracting services/sidebandservice.pyc
03-19 09:10:50.163  1759  4958 W InputManager-JNI: Input channel object 'de20edd Splash Screen io.unsigned.sideband (client)' was disposed without first being removed with the input manager!
03-19 09:10:50.189 27825 27852 V python  : extracting share/mirrors/reticulum.network/manual/_images/sideband_devices.webp
03-19 09:10:50.212 27825 27852 V python  : extracting share/mirrors/unsigned.io/guides/2023_01_19_RNode_Bluetooth_Setup_In_Sideband_On_Android.html
03-19 09:10:50.300 27825 27852 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/gfx/sideband.webp
03-19 09:10:50.308 27825 27852 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/m/_images/sideband_devices.webp
03-19 09:10:50.327 27825 27852 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/s_sideband.html
03-19 09:10:50.330 27825 27852 V python  : extracting share/mirrors/unsigned.io/sideband/
03-19 09:10:50.330 27825 27852 V python  : extracting share/mirrors/unsigned.io/sideband/index.html
03-19 09:10:50.331 27825 27852 V python  : extracting share/mirrors/unsigned.io/software/Sideband.html
03-19 09:10:50.341 27825 27852 V python  : extracting sideband/
03-19 09:10:50.341 27825 27852 V python  : extracting sideband/__init__.pyc
03-19 09:10:50.341 27825 27852 V python  : extracting sideband/core.pyc
03-19 09:10:50.342 27825 27852 V python  : extracting sideband/geo.pyc
03-19 09:10:50.342 27825 27852 V python  : extracting sideband/res.pyc
03-19 09:10:50.342 27825 27852 V python  : extracting sideband/sense.pyc
03-19 09:10:50.349 27825 27852 V pythonutil: Unpacking /data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/lib/arm64/libpybundle app
03-19 09:10:50.349 27825 27852 V pythonutil: Extracting /data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/lib/arm64/libpybundle assets.
03-19 09:10:50.621 27825 27865 V SDL     : Running main function SDL_main from library /data/app/~~RUvQA6yLJRSRafPAKku85w==/io.unsigned.sideband-XSMQ0DSOwHSehpFUZhmRgw==/lib/arm64/libmain.so
03-19 09:10:50.621 27825 27865 I python  : /data/user/0/io.unsigned.sideband/files/app
03-19 09:10:50.622 27825 27865 I python  : /data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip:/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules
03-19 09:10:50.641 27825 27865 I python  : Android path ['.', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/site-packages']
03-19 09:10:50.641 27825 27865 I python  : os.environ is environ({'PATH': '/product/bin:/apex/com.android.runtime/bin:/apex/com.android.art/bin:/system_ext/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'ANDROID_ART_ROOT': '/apex/com.android.art', 'ANDROID_I18N_ROOT': '/apex/com.android.i18n', 'ANDROID_TZDATA_ROOT': '/apex/com.android.tzdata', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'DOWNLOAD_CACHE': '/data/cache', 'BOOTCLASSPATH': '/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/apex/com.android.i18n/javalib/core-icu4j.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar:/system/framework/qcom.fmradio.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar:/system/framework/WfdCommon.jar:/apex/com.android.appsearch/javalib/framework-appsearch.jar:/apex/com.android.conscrypt/javalib/conscrypt.jar:/apex/com.android.ipsec/javalib/android.net.ipsec.ike.jar:/apex/com.android.media/javalib/updatable-media.jar:/apex/com.android.mediaprovider/javalib/framework-mediaprovider.jar:/apex/com.android.os.statsd/javalib/framework-statsd.jar:/apex/com.android.permission/javalib/framework-permission.jar:/apex/com.android.permission/javalib/framework-permission-s.jar:/apex/com.android.scheduling/javalib/framework-scheduling.jar:/apex/com.android.sdkext/javalib/framework-sdkextensions.jar:/apex/com.android.tethering/javalib/framework-connectivity.jar:/apex/com.android.tethering/javalib/framework-tethering.jar:/apex/com.android.wifi/javalib/framework-wifi.jar', 'DEX2OATBOOTCLASSPATH': '/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/apex/com.android.i18n/javalib/core-icu4j.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar:/system/framework/qcom.fmradio.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar:/system/framework/WfdCommon.jar', 'SYSTEMSERVERCLASSPATH': '/system/framework/com.android.location.provider.jar:/system/framework/services.jar:/system/framework/ethernet-service.jar:/apex/com.android.appsearch/javalib/service-appsearch.jar:/apex/com.android.art/javalib/service-art.jar:/apex/com.android.media/javalib/service-media-s.jar:/apex/com.android.permission/javalib/service-permission.jar', 'STANDALONE_SYSTEMSERVER_JARS': '/apex/com.android.os.statsd/javalib/service-statsd.jar:/apex/com.android.scheduling/javalib/service-scheduling.jar:/apex/com.android.tethering/javalib/service-connectivity.jar:/apex/com.android.wifi/javalib/service-wifi.jar', 'ANDROID_SOCKET_zygote': '22', 'ANDROID_SOCKET_usap_pool_primary': '24', 'ANDROID_ENTRYPOINT': 'main.pyc', 'ANDROID_ARGUMENT': '/data/user/0/io.unsigned.sideband/files/app', 'ANDROID_APP_PATH': '/data/user/0/io.unsigned.sideband/files/app', 'ANDROID_PRIVATE': '/data/user/0/io.unsigned.sideband/files', 'ANDROID_UNPACK': '/data/user/0/io.unsigned.sideband/files/app', 'PYTHONHOME': '/data/user/0/io.unsigned.sideband/files/app', 'PYTHONPATH': '/data/user/0/io.unsigned.sideband/files/app:/data/user/0/io.unsigned.sideband/files/app/lib', 'PYTHONOPTIMIZE': '2', 'P4A_BOOTSTRAP': 'SDL2', 'PYTHON_NAME': 'python', 'P4A_IS_WIN [... cut ...]
03-19 09:10:52.238 27870 27885 I sidebandservice: jnius.jnius.JavaException: JVM exception occurred: Class not found using the boot class loader; no stack trace available java.lang.NoClassDefFoundError
03-19 09:10:52.238 27870 27885 I sidebandservice:
03-19 09:10:52.290  1759  2589 I ActivityManager: Process io.unsigned.sideband:service_sidebandservice (pid 27870) has died: prcp FGS
```

**System Information**
- OS: Android 12
- Device: Sony Xperia 1 II
- Sideband 0.7.9

**Additional context**
Working fine a Galaxy A13 running Android 14

markqvist, 2024-03-19:
Thank you very much for this bug report!

If I send you a debug build, with additional logging available, would you be able to capture that with `adb logcat` as well and post it here?

markqvist, 2024-03-19:
When you have a chance, would you mind trying the newly uploaded version 0.8.0 APK on the Xperia device? This build should fail gracefully if WiFiManager is not available, but I am quite interested in seeing the startup logs still, and also in knowing whether the app actually still works on your device, after it has been put to sleep. Can it still receive messages? Or does networking become completely unavailable after a while?

Any further info you can provide would be much appreciated!

Doodiepants, 2024-03-20:
Hi Mark,

Thanks for the fast response et very fast update.

The 0.8.0 build startup nicely.
Here are the logs:
```
03-20 09:16:02.596  1759  5016 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=io.unsigned.sideband/org.kivy.android.PythonActivity bnds=[229,79][442,372]} from uid 10189
03-20 09:16:02.608  1327  1575 E ANDR-PERF-LM: GamePerfEngine: runAsync() 477: Hint is for io.unsigned.sideband/org.kivy.android.PythonActivity/20240214 = 0x1003 PID=1759
03-20 09:16:02.610  5154  5154 I Freeform: onTaskCreated taskId=873 cmp=ComponentInfo{io.unsigned.sideband/org.kivy.android.PythonActivity} target=null
03-20 09:16:02.621  1759  2049 I ActivityManager: Start proc 27386:io.unsigned.sideband/u0a691 for pre-top-activity:com.sonymobile.launcher {io.unsigned.sideband/org.kivy.android.PythonActivity}
03-20 09:16:02.642 27386 27386 D nativeloader: Configuring clns-4 for other apk /data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/base.apk. target_sdk_version=30, uses_libraries=, library_path=/data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/lib/arm64:/data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/base.apk!/lib/arm64-v8a, permitted_path=/data:/mnt/expand:/data/user/0/io.unsigned.sideband
03-20 09:16:02.647 27386 27386 V GraphicsEnvironment: ANGLE Developer option for 'io.unsigned.sideband' set to: 'default'
03-20 09:16:02.721  1759  2030 I ActivityTaskManager: Displayed io.unsigned.sideband/org.kivy.android.PythonActivity: +123ms
03-20 09:16:03.052 27386 27408 V python  : extracting services/sidebandservice.pyc
03-20 09:16:03.080  1759  9909 W InputManager-JNI: Input channel object '96579e6 Splash Screen io.unsigned.sideband (client)' was disposed without first being removed with the input manager!
03-20 09:16:03.106 27386 27408 V python  : extracting share/mirrors/reticulum.network/manual/_images/sideband_devices.webp
03-20 09:16:03.129 27386 27408 V python  : extracting share/mirrors/unsigned.io/guides/2023_01_19_RNode_Bluetooth_Setup_In_Sideband_On_Android.html
03-20 09:16:03.216 27386 27408 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/gfx/sideband.webp
03-20 09:16:03.224 27386 27408 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/m/_images/sideband_devices.webp
03-20 09:16:03.242 27386 27408 V python  : extracting share/mirrors/unsigned.io/rnode_bootstrap_console/s_sideband.html
03-20 09:16:03.245 27386 27408 V python  : extracting share/mirrors/unsigned.io/sideband/
03-20 09:16:03.245 27386 27408 V python  : extracting share/mirrors/unsigned.io/sideband/index.html
03-20 09:16:03.247 27386 27408 V python  : extracting share/mirrors/unsigned.io/software/Sideband.html
03-20 09:16:03.308 27386 27408 V python  : extracting sideband/
03-20 09:16:03.308 27386 27408 V python  : extracting sideband/__init__.pyc
03-20 09:16:03.308 27386 27408 V python  : extracting sideband/core.pyc
03-20 09:16:03.309 27386 27408 V python  : extracting sideband/geo.pyc
03-20 09:16:03.309 27386 27408 V python  : extracting sideband/res.pyc
03-20 09:16:03.309 27386 27408 V python  : extracting sideband/sense.pyc
03-20 09:16:03.317 27386 27408 V pythonutil: Unpacking /data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/lib/arm64/libpybundle app
03-20 09:16:03.317 27386 27408 V pythonutil: Extracting /data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/lib/arm64/libpybundle assets.
03-20 09:16:03.585 27386 27422 V SDL     : Running main function SDL_main from library /data/app/~~EcU6zeuBZUFmOE9ix2hYXQ==/io.unsigned.sideband-LBB4gKw00ir5GgBdCFYOgg==/lib/arm64/libmain.so
03-20 09:16:03.586 27386 27422 I python  : /data/user/0/io.unsigned.sideband/files/app
03-20 09:16:03.586 27386 27422 I python  : /data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip:/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules
03-20 09:16:03.604 27386 27422 I python  : Android path ['.', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/stdlib.zip', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/modules', '/data/user/0/io.unsigned.sideband/files/app/_python_bundle/site-packages']
03-20 09:16:03.604 27386 27422 I python  : os.environ is environ({'PATH': '/product/bin:/apex/com.android.runtime/bin:/apex/com.android.art/bin:/system_ext/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'ANDROID_ART_ROOT': '/apex/com.android.art', 'ANDROID_I18N_ROOT': '/apex/com.android.i18n', 'ANDROID_TZDATA_ROOT': '/apex/com.android.tzdata', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'DOWNLOAD_CACHE': '/data/cache', 'BOOTCLASSPATH': '/apex/com.android.art/javalib/core-oj.jar:/apex/com.android.art/javalib/core-libart.jar:/apex/com.android.art/javalib/okhttp.jar:/apex/com.android.art/javalib/bouncycastle.jar:/apex/com.android.art/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/framework-graphics.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/apex/com.android.i18n/javalib/core-icu4j.jar:/system/framework/tcmiface.jar:/system/framework/telephony-ext.jar:/system [... cut ...]
03-20 09:16:19.627  1327  1575 E ANDR-PERF-LM: GamePerfEngine: runAsync() 477: Hint is for io.unsigned.sideband = 0x1002 PID=1759
03-20 09:16:19.627  1327  1575 W ANDR-PERF-LM: MetaMeter: handleAsync() 171: GameOptimization Invalid state for appName io.unsigned.sideband
```

I can test the debug build if needed.

I'm pretty new to Reticulum and Sideband, s'il need a tiny bit of time to test the messages and the sleeping behaviour.
I'll keep you posted.

Feel free to close the ticket as the app is now starting successfully.

markqvist, 2024-03-20:
Thanks! I'll close this for now, but when/if you have the time, I'm very interested in hearing whether communication over WiFi still works after the phone has been put to sleep.

According to the log, the CPU wakelock is taken correctly, but it can't obtain access to handle the WiFi card. I don't know if it's simply an Xperia thing, where it doesn't have the WiFiManager available in the API, and the wifi card state is managed automatically or something, but hopefully it works as intended still.

closed 2024-03-20 by markqvist