markqvist/Sideband #43: Sideband android app fails to send messages over RNode (LoRA)

VladislavF, 2024-03-26:
**Describe the Bug**
When attempting to send a message through an RNode device (Lilygo T-beam 433MHz SX1278 firmware 1.7) using the Sideband app on android, messages always fail to send going from "Establishing link" to "waiting for path" to "failed". This is evidenced by the fact that my two test devices will begin transmitting simultaneously as seen by the red LED indicator.

**To Reproduce**
Connect two RNode devices with sideband (at least one using the android client). After announcements, attempt to send a message from a device using the android app. The message will fail.

**Expected Behavior**
Sending a message from the android app should complete successfully.

**Logs & Screenshots**
I don't know how to get to log output from Android, I can dump a PC side log.

**System Information**
- Android 13/Ubuntu 22.04
- Python 3.9
- 0.8.1

**Additional context**
This issue is not present when using the Sideband python/pip package. I can run the same radios with two computers (an Ubuntu machine, and an RPi) running the Sideband client without issue. Or even using nomadnet is not a problem either. This issue is somehow specific to the Sideband android app. Similarly, messages sent from the computer to the mobile app will typically succeed. Messages sent from the app always fail.

I have also tried this with the past 3 versions of the app on two different phones.

VladislavF, 2024-03-26:
I should also mention. I point out in the summary, that I can tell when the message is going to fail by the nodes' transmit indicator. The app seems to ignore channel state or something as it will begin sending while the device directly next to it is still transmitting. This again does not happen with the pip version (0.8.0) where the two devices will gracefully ping-pong packets until the message is transferred successfully (without talking at the same time).

markqvist, 2024-03-27:
Hmm, that's odd. Thanks for the report. Can you specify what LoRa settings you used? I don't need the frequency and transmit power, but just the bandwidth, spreading factor and coding rate.

VladislavF, 2024-03-27:
> Hmm, that's odd. Thanks for the report. Can you specify what LoRa settings you used? I don't need the frequency and transmit power, but just the bandwidth, spreading factor and coding rate.

Yes, thank you for reminding me. I meant to post this earlier.
SF: 12
CR: 8
BW: 62.5kHz

I have not tried this with higher rates yet. Thanks for your good work :)

VladislavF, 2024-03-27:
for context these are my exact units: https://www.lilygo.cc/products/t-beam-v1-1-esp32-lora-module?variant=43059202654389

markqvist, 2024-03-27:
Thanks a lot for the detailed info and ideas about the issue. I will try to recreate the problem with the hardware I have here to see if I can replicate and debug it.

Just to be absolutely sure, you linked to the T-Beams based on the SX1278 chip, this is the ones you have, right?

markqvist, 2024-03-27:
Just for info, I'm moving this to the RNode firmware repo. I currently think it's most likely to be firmware related.