markqvist/Sideband #44: Sideband 0.8.1 crashes on startup on Windows

faragher, 2024-03-27:
**Describe the Bug**

Sideband 0.8.1 immediately crashes on startup prior to UI initialization. 

**To Reproduce**

Update Sideband to 0.8.1 (from 0.7.3). Attempt to start from command line using Python 3.8.6. 

**Expected Behavior**

Sideband boots as normal

**Logs & Screenshots**

No screenshot available. Logs are generally useless, but included below.

```
[INFO   ] Logger: Record log in C:\Users\(user)\.kivy\logs\kivy_24-03-26_5.txt
[INFO   ] deps: Successfully imported "kivy_deps.angle" 0.4.0
[INFO   ] deps: Successfully imported "kivy_deps.glew" 0.3.1
[INFO   ] deps: Successfully imported "kivy_deps.sdl2" 0.7.0
[INFO   ] Kivy: v2.3.0
[INFO   ] Kivy: Installed at "c:\users\(user)\appdata\local\programs\python\python38\lib\site-packages\kivy\__init__.py"
[INFO   ] Python: v3.8.6 (tags/v3.8.6:db45529, Sep 23 2020, 15:52:53) [MSC v.1927 64 bit (AMD64)]
[INFO   ] Python: Interpreter at "c:\users\(user)\appdata\local\programs\python\python38\python.exe"
[INFO   ] Logger: Purge log fired. Processing...
[INFO   ] Logger: Purge finished!
```

**System Information**

Windows 10
Python 3.8.6
Sideband 0.8.1
RNS 0.7.3

**Additional context**

I'm hoping there are other logs I am unaware of.

markqvist, 2024-03-27:
In [main.py](https://github.com/markqvist/Sideband/blob/main/sbapp/main.py), can you edit the first line to be `__debug_build__ = True`, and try launching again to see if there is more output?

I'm not exactly sure where Windows stores installed Python packages, but it should be possible to find it under your user directory somehwere - maybe look at the `PATH` environment variable.

faragher, 2024-03-27:
At least for me it's AppData\Local\Programs\Python\Python38\Lib\site-packages

I think I see the issue: 

`[WARNING] stderr: OSError: File C:\Program Files/sbapp/assets/fonts/NotoSansHebrew-Regular.ttf not found`

Looks like a Windows specific directory issue. By copying the fonts to the Program Files directory, the program functions correctly. I'll see if I can find what shortcut/alias is causing the issue.

markqvist, 2024-03-27:
Aha! That is most likely it. These directory/data/assets have been the bane of my existence. It's different methods on all platforms, and varies depending on installation type and whether it's run from source or from an installed package/distribution.

Since the [font loader](https://github.com/markqvist/Sideband/blob/main/sbapp/main.py#L366) uses the `asset_dir` property on the Sideband core instance, this should in theory work if we add correct handling for Windows to [the logic that finds](https://github.com/markqvist/Sideband/blob/main/sbapp/sideband/core.py#L147) the `asset_dir`.

As you can see in that code, there's already sections for Linux, Android and macOS, so if we can figure out the correct way to universally get this on Windows, and add a case for that, that should solve it on Windows for this and any other future assets. Any help here would be **much** appreciated, since I don't have access to any Windows machines/VMs right now.

faragher, 2024-03-27:
Windows 10 works with the following addition to core.py:158

```
        elif RNS.vendor.platformutils.is_windows():
            core_path          = os.path.abspath(__file__)
            self.asset_dir     = core_path.replace("\sideband\core.py", "/assets")
```

I would like to point out  `self.asset_dir     = core_path.replace("\sideband\core.py", "\assets")` does not work. Escape sequence issue, most likely. Windows is pretty lax with \ vs / these days, but I don't know how far back that goes. Of course, you could also escape it with: 

```
        elif RNS.vendor.platformutils.is_windows():
            core_path          = os.path.abspath(__file__)
            self.asset_dir     = core_path.replace("\sideband\core.py", "\\assets")
```

which may or may not be more compatible. I can't bring my XP machine up to speed to run RNS, and it probably doesn't have the RAM anyway, and I don't have a Win7 machine to test on, but for 10 it appears to work.

markqvist, 2024-03-27:
Thanks a bunch for figuring that one out and testing it!

Ok, that's great. I think we'll go with the second one, since that seems to be the right "Window-ish" way of having the slashes, even though it does kinda give me the shivers ;)

One thing that really confuses me here though: Does Python on Windows not see the first part of `\sideband` as an escape sequence? Are we completely sure it should not be `\\sideband`?

faragher, 2024-03-27:
Why does it work? No clue. I'm with you that either it shouldn't be required at all, or it should be required for both. 

Without the escape we get `appdata\local\programs\python\python38\lib\site-packages\sbappssets/fonts/NotoSansHebrew-Regular.ttf` 

If we escape all of them (`core_path.replace("\\sideband\\core.py", "\\assets")`) then it still works. I wonder if replace has some kind of auto-escape on input but not output . . .

If it works, and it makes sense, and it's consistent, then escaping all of them might be the best practice. That brings the final tested modification to: 

```
        elif RNS.vendor.platformutils.is_windows():
            core_path          = os.path.abspath(__file__)
            self.asset_dir     = core_path.replace("\\sideband\\core.py", "\\assets")
```

markqvist, 2024-03-27:
Haha, okay, we may never know. I think we'll run with that version then :)

markqvist, 2024-03-27:
Fixed by 48908842c6c98a9e9aa2c1c984dd76c66fca3300.

closed 2024-03-27 by markqvist

markqvist, 2024-03-27:
I hotfixed this one into version 0.8.1, so there is a new wheel on `pip` now that should work without modifications on Windows :)