markqvist/Sideband #45: MMAP API varies between OS types. Non-functional on Windows

faragher, 2024-03-28:
**Describe the Bug**

MMAP in GeoID fails gracefully, allowing function but preventing accessing the telemetry from being displayed as text when selecting a node. (telemetry screen is blank)

**To Reproduce**

Select map marker on Situation Map

**Expected Behavior**

Telemetry properly displayed.

**Logs & Screenshots**

Not provided, solution below.

**System Information**
Windows 10
Python 3.8
Sideband 0.8.1

**Additional context**

MMAP has a different API for Windows vs Unix. https://docs.python.org/3/library/mmap.html#mmap.mmap

Solution is to edit geo.py: ~421 to read:

```
            if RNS.vendor.platformutils.is_windows():
                self.raw = mmap.mmap(fd, fullsize, access=mmap.ACCESS_READ)
            else:
                self.raw = mmap.mmap(fd, fullsize, mmap.MAP_SHARED, mmap.PROT_READ)
```
Tested and functional on Windows 10.

I cannot, for the life of me, get it to run from the repo. I'll try to make a pull request, but I had to modify my package, and I can't test in-situ.

markqvist, 2024-03-28:
Great, thanks for spotting and fixing this! Merged and working fine on Unix too, as expected.

I compiled a pre-release test wheel with these changes included that's [available for download here](https://unsigned.io/dl/sbapp-0.8.2-py3-none-any.whl).

If you can test that the telemetry display (and anything else you have time for testing) works on Windows with that, it would be great.

faragher, 2024-03-28:
I'll do testing and look into making an .exe tomorrow. I know there are people that'd really appreciate having a standalone executable. 

That being said . . . I don't have another Windows system functioning right now. I think I can get an old laptop running again (HP makes fragile garbage, don't buy it. Asus laptops are the only ones I recommend) but it may take a bit. I don't think it has Pip or maybe even Python, so we can start from zero.

markqvist, 2024-03-28:
That is absolutely awesome! Thanks so much for taking a shot at it. I don't think we need to target anything earlier than Windows 10, if it makes things less complicated, but given backwards compat is usually pretty simple on Windows, it would definitely wouldn't hurt if it worked on 7 as well, even though I have no idea if people use that anymore. Actually, you probably be the judge of that - I have no idea what I'm talking about.

Hah, I've had completely the same experience with HP and Asus ones. The HPs fall apart after being looked at wrong, but once left an Asus laptop in the garden for three days of intermittent rain, and it somehow survived, and continued to operate for several more years, until it was put into a spare locker. It probably still works fine.

faragher, 2024-03-28:
There's iconoclasts who still run 7. I'd still be running 7 if Cyberpunk 2077 didn't force me to upgrade. No, really. I liked it better anyway.

I can't match rain, but my Asus was thrown into a backpack daily for several years. The entire side of plastic is smashed out and the monitor's ribbon cable is damaged so you need to position it *just right* for it to work, but it's an external display away from being perfect. First x64 and widescreen machine I owned. *That* was a learning curve.

I'm not a fan of Windows for heavy lifting (it's more gaming in my eyes) but it's widely used by consumers and businesses, so it's helpful to make it compatible if possible. I only know of one other error at this time, but that may be firewall related, so it's not a representative test.

faragher, 2024-03-29:
0.8.2 wheel starts and displays telemetry correctly, so it looks like it fixes the issues I know about. 

Have confirmed the tilemap issue was my firewall.

faragher, 2024-03-29:
Progress report: PyInstaller wasn't working as-is. Missing a lot of dependencies. Likely my configuration being incorrect.

Found Auto Py to Exe, which is a frontend that takes care of a lot more of the work, apparently. It's been doing a trace and finding dependencies for three hours. If this is a one-time thing, and new builds are just running this new script, that's fine, but if this is going to be a multi-hour process for every build, just to avoid pip/python, then we need to consider other solutions.

I'll keep you in the loop.

faragher, 2024-03-29:
I've drilled down a ways, and got it to the point where I'm getting conceptual errors. I'm getting a plyer NotImplemented error. Since the program works without being compiled, I'm confident in saying a PyInstaller version would require considerable tweaks and modifications. Or at least someone more familiar with the process. I'll keep an eye open for other ways.

markqvist, 2024-03-29:
Ok, if it's getting that complicated and takes so long to run, I think we might just stick with the pip builds for now. I'll have a look at it occasionally as well, so if you do come across any more info or hints, you're very welcome to share them here, or in a new thread.

Since it seems to be all working now, I'll close this issue and put out a new release build with the Windows errors fixed later today.

Thanks for finding and fixing it!

Btw, if you enjoyed the main Cyberpunk 2077, did you play and find the new expansion enjoyable too? I was wondering if it was worth giving it a second spin for the expansion.

closed 2024-03-29 by markqvist

faragher, 2024-03-29:
It's a hard call. It's exceptionally well put together as a story, I mean, there's a scene where I'm in a car with a guy, and it's detailed enough you can actually see what he's thinking with just where he's looking and his body language. It's everything they did right with the acting and animation, but turned to 11.

It feels a bit like a spy thriller, and the story is pretty decent, with a few really exceptional moments. I could say the same about CP77 in general.

There is one . . . polarizing . . . gameplay segment in one of the paths for the main quest. I hated it. Just hated it. Still, if you haven't played since they retooled Cyberware, it's like playing a new game, mechanically. For better and for worse.

I don't have your wallet or your schedule, so I don't know how those things factor in, but I generally think it's worth playing. But I think my real review is: I still have it installed, I have 245 hours in the game, I've played Phantom Liberty two complete times, I think, and after talking about it, I kinda want to play it again.

faragher, 2024-03-29:
Oh, and that build failed. Probably just a bad build. Using PyInstaller, I'm still unable to bypass the plyer issue, despite it being a somewhat known issue. I don't know what's going on with it.

I still prefer pip, anyway. I see the appeal of a contained program, though.

markqvist, 2024-03-29:
Thanks, that's as much as a recommendation I needed :) I really liked the original story, the way it was told and presented, and how you interacted with it. And no, I put it down just before that update, so that's nice to hear. Since it's been a while now, it'll probably be fine as a new, but slightly familiar experience, even though I normally dislike when someone comes in and changes the gameplay mechanics I've figured out and gotten used to ;)