markqvist/Sideband #51: Cannot connect RAK4631-based RNode to Sideband via USB

jacobeva, 2024-07-05:
**Describe the Bug**
Cannot connect RAK4631-based RNode to Sideband via USB.

**To Reproduce**
* Enable RNode in Sideband hardware settings and configure it
* Restart sideband
* RNode is never connected to

**Expected Behavior**
The RNode should be connected to by Sideband and used as a communication device.

**Logs & Screenshots**
On startup:
```
07-05 22:37:02.894 20868 20887 I sidebandservice: [2024-07-05 22:37:02] [Debug] Adding RNode Interface...
07-05 22:37:02.894 20868 20887 I sidebandservice: [2024-07-05 22:37:02] [Debug] Disallowing RNode bluetooth since config is disabled
07-05 22:37:02.895 20868 20887 I sidebandservice: [2024-07-05 22:37:02] [Error] Could not open serial port for interface RNodeInterface[RNodeInterface]
07-05 22:37:02.895 20868 20887 I sidebandservice: [2024-07-05 22:37:02] [Error] The contained exception was: Neither serial port nor Bluetooth devices available
07-05 22:37:02.895 20868 20887 I sidebandservice: [2024-07-05 22:37:02] [Error] Reticulum will attempt to bring up this interface periodically
```

When I try to enable bluetooth in the RNode section of hardware settings:
```
07-05 22:29:39.849 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3-libbz2/arm64-v8a__ndk_target_24/python3/Lib/threading.py", line 1016, in _bootstrap_inner
07-05 22:29:39.857 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/sideband/arm64-v8a/jnius/__init__.py", line 85, in jnius_thread_hook
07-05 22:29:39.858 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3-libbz2/arm64-v8a__ndk_target_24/python3/Lib/threading.py", line 953, in run
07-05 22:29:39.858 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/app/sideband/core.py", line 2684, in _service_jobs
07-05 22:29:39.859 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/app/RNS/Interfaces/Android/RNodeInterface.py", line 671, in enable_bluetooth
07-05 22:29:39.859 16603 16633 I sidebandservice:    File "/home/markqvist/Information/Source/Sideband/sbapp/.buildozer/android/app/RNS/Interfaces/Android/RNodeInterface.py", line 515, in write_mux
07-05 22:29:39.860 16603 16633 I sidebandservice:  OSError: No ports available for writing
```

**System Information**
- Fairphone 3, Android 13
- Sideband v0.8.5

**Additional context**
This is the same bug I mentioned a while ago in the Matrix channel briefly, I have now collected relevant logs. Am I actually configuring this correctly? Are there any dependencies I should be installing on my phone?
All permissions except location are enabled in settings for the app.

jacobeva, 2024-07-06:
Is there any way I can change the RNS log level in Sideband? I have a feeling you forgot to add the RAK4631 to the device filter (here)[https://github.com/markqvist/Sideband/blob/7ab94e81ea00ce680de5ed25f8da2d1470e5e29e/sbapp/services/sidebandservice.py#L54]. But I can't see [this](https://github.com/markqvist/Sideband/blob/7ab94e81ea00ce680de5ed25f8da2d1470e5e29e/sbapp/services/sidebandservice.py#L244) line in my logcat at the moment.

jacobeva, 2024-07-08:
Fix created. See PR https://github.com/markqvist/Sideband/pull/52

jacobeva, 2024-07-08:
Best leave it open until it's actually merged.

imBunyip, 2024-08-08:
while this is still open ,
i have the same error using sideband on android with the new lilyGo T3S3 version 1.2.
if it could be added too, the T3S3 comes with  pid:0x1001 and vid:0x303a 

`
class SidebandService:
    usb_device_filter = {
        0x0403: [0x6001, 0x6010, 0x6011, 0x6014, 0x6015], # FTDI
        0x10C4: [0xea60, 0xea70, 0xea71], # SiLabs
        0x067B: [0x2303, 0x23a3, 0x23b3, 0x23c3, 0x23d3, 0x23e3, 0x23f3], # Prolific
        0x1a86: [0x5523, 0x7523, 0x55D4], # Qinheng
        0x0483: [0x5740], # ST CDC
        0x2E8A: [0x0005, 0x000A], # Raspberry Pi Pico
        0x239A: [0x8029], # Adafruit (RAK4631)
        0x303A: [0x1001], # T3S3 LilyGO ESP32-S3
    }
`

markqvist, 2024-08-17:
Merged now and closing.

@imBunyip: Added in bbefc39fd79001dddf5dc11c09a4fcf3b9b4948b :)

Release coming relatively soon :)

closed 2024-08-17 by markqvist