markqvist/Sideband #54: Touchpad inputs are treated as touchscreen inputs on Linux under some unknown conditions

felurx, 2024-08-17:
**Describe the Bug**
Sideband or Kivy seem to partially treat inputs from a touchpad as if they were coming from a touchscreen on my system, which results in the GUI being pretty much unusable via touchpad. 

For example, moving a finger up and down on a scrollable view results in its contents moving up and down (instead of just moving the mouse pointer). Also, tapping on the touchpad seems to sometimes click on things corresponding to where the touchpad was tapped, instead of where the mouse pointer is.
There is a screencast below.

Using a regular mouse works fine, even without closing and reopening Sideband.

**To Reproduce**
I don't know which pieces of my system cause the bug to surface. A possible starting point might be to live-boot Manjaro with KDE from a USB stick on a laptop and install Sideband on that (via pipx).
After that, just opening Sideband and using the touchpad should be enough to see the bug in action.

**Expected Behavior**
Sideband should treat touchpads like other mouse input devices.

**Logs & Screenshots**
<details><summary>Sideband log file with -v</summary>
<p>

```
[INFO   ] Logger: Record log in /home/felix/.kivy/logs/kivy_24-08-17_4.txt
[INFO   ] Kivy: v2.3.0
[INFO   ] Kivy: Installed at "/home/felix/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/kivy/__init__.py"
[INFO   ] Python: v3.12.4 (main, Jun  7 2024, 06:33:07) [GCC 14.1.1 20240522]
[INFO   ] Python: Interpreter at "/home/felix/.local/share/pipx/venvs/sbapp/bin/python"
[INFO   ] Logger: Purge log fired. Processing...
[INFO   ] Logger: Purge finished!
[INFO   ] KivyMD: 1.2.0.dev0 (installed at "/home/felix/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/sbapp/kivymd/__init__.py")
[INFO   ] Factory: 195 symbols loaded
[DEBUG  ] Cache: register <kv.resourcefind> with limit=None, timeout=60
[DEBUG  ] Cache: register <kv.image> with limit=None, timeout=60
[DEBUG  ] Cache: register <kv.atlas> with limit=None, timeout=None
[INFO   ] ImageLoaderFFPy: Using ffpyplayer 4.5.1
[INFO   ] Image: Providers: img_tex, img_dds, img_sdl2, img_pil, img_ffpyplayer 
[DEBUG  ] Cache: register <kv.texture> with limit=1000, timeout=60
[DEBUG  ] Cache: register <kv.shader> with limit=1000, timeout=3600
[DEBUG  ] Cache: register <kv.graphics.texture> with limit=None, timeout=None
[INFO   ] Text: Provider: sdl2
[DEBUG  ] Cache: register <kv.lang> with limit=None, timeout=None
[INFO   ] Window: Provider: sdl2
[INFO   ] GL: Using the "OpenGL" graphics system
[INFO   ] GL: Backend used <sdl2>
[INFO   ] GL: OpenGL version <b'4.6 (Compatibility Profile) Mesa 24.1.5-manjaro1.1'>
[INFO   ] GL: OpenGL vendor <b'AMD'>
[INFO   ] GL: OpenGL renderer <b'AMD Radeon 760M (radeonsi, gfx1103_r1, LLVM 18.1.8, DRM 3.54, 6.6.44-1-MANJARO)'>
[INFO   ] GL: OpenGL parsed version: 4, 6
[INFO   ] GL: Shading version <b'4.60'>
[INFO   ] GL: Texture max size <16384>
[INFO   ] GL: Texture max units <32>
[DEBUG  ] Shader: Fragment compiled successfully
[DEBUG  ] Shader: Vertex compiled successfully
[DEBUG  ] ImageSDL2: Load </home/felix/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/kivy/data/glsl/default.png>
[INFO   ] Window: auto add sdl2 input provider
[INFO   ] Window: virtual keyboard not allowed, single mode, not docked
[INFO   ] Clipboard: Provider: xclip
[INFO   ] CutBuffer: cut buffer support enabled
[INFO   ] SoundFFPy: Using ffpyplayer 4.5.1
[DEBUG  ] Audio: register SoundFFPy
[DEBUG  ] Audio: register SoundSDL2
[DEBUG  ] Audio: register MusicSDL2
[INFO   ] Audio: Providers: audio_ffpyplayer, audio_sdl2 
[DEBUG  ] Cache: register <textinput.label> with limit=None, timeout=60.0
[DEBUG  ] Cache: register <textinput.width> with limit=None, timeout=60.0
[DEBUG  ] App: Loading kv </home/felix/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/sbapp/sideband.kv>
[DEBUG  ] App: kv </home/felix/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/sbapp/sideband.kv> not found
[DEBUG  ] Resource: add </usr/share/fonts> in path list
[DEBUG  ] Resource: add </usr/share/fonts/cantarell> in path list
[DEBUG  ] Resource: add </usr/share/fonts/noto> in path list
[DEBUG  ] Resource: add </usr/share/fonts/gsfonts> in path list
[DEBUG  ] Resource: add </usr/share/fonts/rnote-fonts> in path list
[... 154 lines cut ...]
Rotation:         n/a

Device:           PIXA3854:00 093A:0274 Touchpad
Kernel:           /dev/input/event18
Group:            9
Seat:             seat0, default
Size:             111x73mm
Capabilities:     pointer gesture
Tap-to-click:     disabled
Tap-and-drag:     enabled
Tap drag lock:    disabled
Left-handed:      disabled
Nat.scrolling:    disabled
Middle emulation: disabled
Calibration:      n/a
Scroll methods:   *two-finger edge 
Click methods:    *button-areas clickfinger 
Disable-w-typing: enabled
Disable-w-trackpointing: enabled
Accel profiles:   flat *adaptive custom
Rotation:         n/a
```

</p>
</details>

felurx, 2024-08-17:
In Kivy on Linux, multitouch devices (= touchscreens) seem to be handled by [kivy.input.providers.mtdev](https://kivy.org/doc/stable/api-kivy.input.providers.mtdev.html), which in the default config is configured by [kivy.input.providers.probesysfs](https://kivy.org/doc/stable/api-kivy.input.providers.probesysfs.html).

This lead me to a workaround:
Commenting out line 49 `%(name)s = probesysfs` under `[input]` in `~/.kivy/config.ini` (which is auto-created by Kivy) disables that behaviour, the touchpads are no longer treated as multitouch devices and everything is fine for me.
(This would probably break actual touchscreens, however.)

I'll investigate further, but I suspect that this is a problem with ProbeSysFs' logic for how to detect touchscreens and should be fixed upstream in Kivy. Or does Sideband do anything unusual regarding input devices?

felurx, 2024-08-17:
To be more precise and correct:
The term "multitouch device" seems to apply to all devices that can track multiple touch points, which includes both touchscreens and touchpads. Also, both have the properties that ProbeSysfs checks for.

The problem seems to be that all multitouch devices are treated as touchscreens, where device coordinates map to screen coordinates somehow and a contact moving corresponds to the user dragging UI elements.
Touchpads should probably just be ignored, since Kivy already deals with mouse events just fine and the OS does everything to create these from touchpad input.

I don't know where that conflation of touchscreens and touchpads happens and how it should be fixed. Also, the question remains why this didn't cause any trouble for other users.

markqvist, 2024-08-17:
Thanks for collecting all the information and posting detailed logs and info. I agree with your observations and initial conclusions here. Unfortunately, Sideband doesn't do anything non-standard in regards to input, so this may be an issue that has it's root in Kivy itself. Regardless, we'll need to find some sort of tenable workaround as well, until it can be identified and potentially fixed in Kivy.

I'll start looking into this, and seeing if I can track down where the issue might occur exactly, but it might be hard for me to debug, since I doubt I have any hardware where I can replicate. Therefore, any additional information and input you can come up with in the meantime is highly appreciated!

Rick1029, 2025-09-27:
I have this same issue, and its persisted from version 1.5 to 1.7 of Sideband. I'm running standard Ubuntu 22.04 on a laptop with a touchpad, and inputs are interpreted as a touchscreen, and keeping a finger in place is treated like you are holding a button on a touchscreen (so a menu item gets highlighted / false clicked). The interface is effectively unusable because of this. 

Commenting out line 49 in ~/.kivy/config.ini as felurx said is a perfect fix and sideband becomes usable on  a laptop. Maybe a menu toggle or popup on startup informing the user of the potential issue with a button to comment out the line (and a menu option in settings to reverse it should it be problematic) if a device with a touchpad is detected could be useful? Although I have no idea how it would break other Kivy reliant applications (if at all).

closed 2026-10-01 by markqvist