markqvist/Sideband #55: Auto Correct not working on Samsung S24

liamcottle, 2024-08-18:
**Describe the Bug**

When writing messages in Sideband for Android, autocorrect is not working.

I saw this issue https://github.com/kivy/python-for-android/pull/2692 that mentioned it was resolved in `2.26.0`, although I only see a kivy dependency and not a `python-for-android` dependency in setup.py of Sideband.

According to this comment https://github.com/kivy/kivy/issues/7337#issuecomment-1281083321 it's suggested that `input_type: text` should be added to the `TextInput`. I do note that Sideband is using `MDTextField`, but I would assume this would get passed up the chain..?

I haven't attempted to build Sideband myself as I believe there's a lot of moving parts, but it would be awesome if you get a chance to look into this :)

**To Reproduce**

- Install Sideband 0.9.1 on Android (Samsung S24, Android 14)
- Open a conversation and start typing a message
- Observe that there are no suggestions, nothing is auto corrected, and messages come out funky unless you type carefully.

**Expected Behavior**

Text is auto corrected when writing a message.

**Logs & Screenshots**

https://github.com/user-attachments/assets/d2b60c9e-a325-40aa-b41f-957c8d611e89

**System Information**

- OS and version: Android 14 on Samsung S24
- Python version: ?
- Program version: 0.9.1

markqvist, 2024-08-30:
The problem with adding `input_type: "text"`, is that backspace / deleting characters then stops working in the text field:

https://github.com/user-attachments/assets/71b5ef06-4e5c-4096-af2f-b00044bcb713

It's a super annoying and really hard to track down bug, and you have no idea how many hours I have spent on it already, and I haven't been able to find reliable solution that works across devices. If figure out something that actually works, I am very interested in hearing it! :)

brothercorvo, 2024-09-21:
same issue on Android 14

markqvist, 2024-09-21:
> same issue on Android 14

Did you read the above? The relevant property on the text field is **intentionally** disabled so that deleting text will work.

There is an underlying bug in Kivy/KivyMD somwhere.

commit 919cd6a4 (2024-09-22): Allow predictive text on Android by default, add option to block predictive text. Fixes #55.
commit 26899f8c (2024-09-22): Allow predictive text on Android by default, add option to block predictive text. Fixes #55.
closed 2024-09-22 by markqvist