markqvist/Sideband #56: No notification when messages are received

Skretch1974, 2024-08-19:
Hi. There is a problem of not having any notification when receiving a message on Android version Sideband β 0.9.1
I'm testing Windows and Android programs, the messages themselves are delivered without problems, I can see that my message sent from Windows is delivered successfully, but on my smartphone there is no indication that the message is received. If I unlock the screen and go into the app, I can see that the message was successfully received. All permissions are granted, all notification checkboxes are enabled, but there is still no indication that the message was received.
I am using a custom firmware version without Google services on Android 12.

markqvist, 2024-08-30:
Is the contact/object marked as trusted? Currently, Sideband will only post notifications for trusted peers.

Skretch1974, 2024-08-31:
Hi. Yes I have marked all my added contacts as trusted.
Today I downloaded and updated the program version to v.0.9.2 but unfortunately neither I nor people from my contacts have any notifications about incoming messages. 
There are not even any changes in the top part of the screen under the curtain, where the emblem of the running Sideband application hangs in the background. 
If I go into the program, I see an envelope icon in the Conversations list, but there are no other external signs of incoming messages.

Skretch1974, 2024-09-21:
Hi. A few comments and suggestions on the application. 
1. It is necessary to prohibit playback of the incoming voice message at the moment when the Ptt button is pressed. The simplex communication mode is required. At the moment the message starts playing at the moment of dictation when the Ptt button is pressed and this leads to superimposition of two sounds on the outgoing message.
2. On tested several different versions of android smartphones from 8 to 14, none of them when receiving text messages have any audible or visual signals about the incoming message. You have to fully open the program window itself and only in it you can see the incoming message. Suggestion - when receiving a message to change the emblem at the top of the screen to the sign of the envelope and need to form a sound beeper signal like it was in the application ICQ early versions with the ability to independently replace the audio file beeper in the application folder. Ideal option to use the internal messaging service of the android itself, as is done in other messengers, such as monocles chat (xmpp jabber).
3. If possible, add a mode to autoload the application when the smartphone is turned on or rebooted.
As we test it, I will add the identified problems in the work of the application. Regards to you.

markqvist, 2024-09-21:
Are you completely certain that you actually enabled notifications in the Preferences section of Sideband? It is turned off by default.

> At the moment the message starts playing at the moment of dictation when the Ptt button is pressed and this leads to superimposition of two sounds on the outgoing message.

What version are you running? This was fixed in one of the recent releases. Incoming audio messages are queued, and will not play until PTT button is released.

> On tested several different versions of android smartphones from 8 to 14, none of them when receiving text messages have any audible or visual signals about the incoming message.

Sideband uses the native notifications system of the operating system, including on Android. This was tested and working on Android 8.1 through 14. Did you enabled notifications in settings, allow notification permission to the app in Android settings? Also, SIdeband will only post notifications for trusted contacts - if a contact is not marked as trusted, notifications will not be posted to the OS.

In Android you can configure notification sounds, vibration and such in the Android notification settings. This can even be individualised for each conversation if you want to.

markqvist, 2024-09-21:
On a second inspection, there is indeed something fishy going on with notifications on some Android distributions. I'll investigate.

Skretch1974, 2024-09-21:
[image]
[image]
[image]
Version 0.9.7.
Yes everything is enabled and in the program in the settings and in the notifications section of the android itself. Absolutely all possible checkboxes of notification settings are enabled, but there are no vibration or sound notifications.
Today several times while pressing the Ptt key and dictating a message, the loudspeaker from the received message turned on and the sound overlapped. 

Translated with DeepL.com (free version)

brothercorvo, 2024-09-21:
this is a confirmed bug with Android 14
[image]
permission are allowed in the OS

[image]

and active in Sideband
[image]

closed 2024-09-21 by markqvist

markqvist, 2024-09-22:
There is now a hotfixed APK available that solves this issue here: https://github.com/markqvist/Sideband/releases/tag/0.9.7

liamcottle, 2024-09-26:
Can confirm Sideband now shows notifications for me on Android 14, Samsung Galaxy S24 after updating to newer version :) Thanks again!

markqvist, 2024-09-26:
Nice, good to hear!

commit 4a12c136 (2024-09-21): Set immutable flag on notification intent. Closes #56.