markqvist/Sideband #58: gps not working.

michael23505, 2024-09-21:
**Read the Contribution Guidelines**
Before creating a bug report on this issue tracker, you **must** read the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md). Issues that do not follow the contribution guidelines **will be deleted without comment**.

- The issue tracker is used by developers of this project. **Do not use it to ask general questions, or for support requests**.
- Ideas and feature requests can be made on the [Discussions](https://github.com/markqvist/Reticulum/discussions). **Only** feature requests accepted by maintainers and developers are tracked and included on the issue tracker. **Do not post feature requests here**.
- After reading the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md), delete this section from your bug report.

**Describe the Bug**
sideband on android has been unable to gather location data from 3 different android phones, but fixed position and other telemetry work as expected.

**To Reproduce**
attempt to acquire and send or receive location telemetry of an android phone's gps using the sideband app.

**Expected Behavior**
location data from a phone's gps should be gathered by the app with telemetry and location settings turned on. this data should then be shared with other users, and displayed on the built in map as expected.

**Logs & Screenshots**
Please include any relevant log output. If applicable, also add screenshots to help explain your problem.

**System Information**
samsung galaxy a3, a11, and a52,
android version 14
sideband app versions 0.9.5 beta and 0.9.7 beta

**Additional context**
Add any other context about the problem here.

markqvist, 2024-09-21:
Have you set permission location to "Allow all the time" for Sideband in Android permissions settings? For telemetry, location is collected by the background service, which needs this permission to actually get the GPS location.

brothercorvo, 2024-09-21:
I can confirm this bug. this is not a permission issue: for some reason the position is provided when the device in NOT moving. when it's moving the position returns as a void. We debugged this in [Reticulum-Telemetry-Hub](https://github.com/FreeTAKTeam/Reticulum-Telemetry-Hub) several times

markqvist, 2024-09-21:
> I can confirm this bug. this is not a permission issue: for some reason the position is provided when the device in NOT moving. when it's moving the position returns as a void. We debugged this in [Reticulum-Telemetry-Hub](https://github.com/FreeTAKTeam/Reticulum-Telemetry-Hub) several times

As far as I can read, OP is referring to *setting a fixed position* in the telemetry settings, not anything about non-stationary devices "returning a void".

Please don't conclude that "this is not a permission issue" on behalf of someone else; that is just hijacking OPs thread.

If you have further information about the issue you're describing, and ideally `adb logcat` output with debug log turned on, please open a separate issue. I'd be very interested to look into it, because I have not seen anything like that across *many* different devices, both moving and stationary.

michael23505, 2024-09-21:
didn't mean to cause confusion. it's my first post anywhere, and i've only been using sideband for a couple days. i just wanted to see if anyone else had the gps problem or if it was being worked on. i really like sideband and what it is becoming capable of, and i'd love to see it fully functional soon.

markqvist, 2024-09-22:
No worries @michael23505, in the reply I above I was specifically answering brothercorvo. Still waiting for your reply on this:

> Have you set permission location to "Allow all the time" for Sideband in Android permissions settings? For telemetry, location is collected by the background service, which needs this permission to actually get the GPS location.

Sideband gets the GPS information via a background service on Android, so that it can update telemetry data even while the app is in the background. This means that you need to allow it to "Access location anytime" though, and since it's not possible on Android to *ask* a user for this permission directly, you must manually go into settings on your Android device and enable this permission. It's a bit contrived, but there's currently no other way to effectively get location data on Android in a reliable way that will work for telemetry purposes.

Can you confirm whether or not this permission has been granted, and whether or not it still does not work when it has?

michael23505, 2024-09-22:
location permission is set to allow all the time on each device. still, none are working.

markqvist, 2024-09-22:
Thanks.

- Can you please describe your setup and what steps you are taking to send the telemetry?
- Is other telemetry included when enabled?
- Can you see your own location on the map by going to the map screen and pressing the "locate me" button?
- Can you see your own coordinates in "own telemetry"?

markqvist, 2024-09-22:
Also, what Android versions are these phones running. Are all of them on Android 14?

michael23505, 2024-09-22:
a samsung android phone with sideband sending to a windows desktop with the python version of sideband over a local wifi, and tcp to the amsterdam test server. windows sideband set as collector. i enabled battery telemetry and it works fine. just no gps no matter what i try. the phone location shows up on the phone's sideband map. but no actual coordinates are visible in telemetry or sent to any other device. samsung galaxy a52 with android 14, galaxy a03 with android 13, and galaxy a11 with android 13.

markqvist, 2024-09-22:
Thanks. If I provide you with a debug build, can you send me log output of `adb logcat | grep "python\|sidebandservice"`?

michael23505, 2024-09-22:
sure.

markqvist, 2024-09-22:
Nice, here's a debug build with extra logging output:

https://github.com/markqvist/Sideband/releases/download/0.9.6/sideband-0.9.7-debug.apk

Please make sure to also allow a minute or two for the service to actually acquire a location update from the OS, as the interval at which this is initially sent can vary quite a bit. Once you see location in "Own Telemetry" and on the map, it should be included in any outgoing telemetry, as long as "Send Telemetry" is enabled in the conversation settings, of course.

michael23505, 2024-09-22:
using debug build on one phone. location still not working. how do i get to the log?

markqvist, 2024-09-22:
Install `adb` on a computer, enable USB debugging on the phone, run:

`adb logcat | grep "python\|sidebandservice"` (or similar)

michael23505, 2024-09-22:
adb won't install. it keeps stopping at failed building wheel for m2crypto.

michael23505, 2024-12-12:
Thank you for the built in log viewer in sideband version 1.2.0. Made this so much easier. My logs say "An error occurred while packing location sensor data. The contained exception was: argument out of range." I hope this helps narrow down the problem.

markqvist, 2024-12-13:
That's very odd, not where I expected it to happen. I've been trying to reproduce it, but can't.

At this point, I'll need to see the raw sensor data coming from the GPS to determine what is going on.

I've created a debug build that will write the raw values to the log just before it fails. Please note, this will write your GPS location to the log! If you don't want to post that publicly on the web, you should probably not include it here on this thread, but you can send it to me in an LXMF message to me at `8dd57a738226809646089335a6b03695`.

If you can just copy and paste the entire log into a message to me after it fails, I should definitely be able to figure out what is going wrong.

You can download the debug APK file here: https://github.com/markqvist/Sideband/releases/download/0.9.6/sideband-1.2.0-gps_debug.apk

markqvist, 2024-12-13:
Thanks for the info. I've implemented a potential fix. Can you try this new build and see if it works now?

https://github.com/markqvist/Sideband/releases/download/0.9.6/sideband-1.2.0-gpstest_1.apk

michael23505, 2024-12-13:
You are a genius. That appears to have fixed it.

commit 426c9d96 (2024-12-13): Updated GPS data struct packing. Fixes #58.
closed 2024-12-13 by markqvist