markqvist/Sideband #61: Fresh install of Sideband 1.1.3 on macOS Sonoma 14.4.1

DeVLaB4U, 2024-11-07:
### I have issue installing Sideband 1.1.3 on macOS Sonoma 14.4.1

`pipx install "sbapp[macos]"`

Fatal error from pip prevented installation. Full pip output in file:
    /Users/tst/.local/pipx/logs/cmd_2024-11-07_03.54.39_pip_errors.log

**pip seemed to fail to build package:
    pycodec2**

Some possibly relevant errors from pip install:
    error: subprocess-exited-with-error
    ValueError: 'pycodec2/pycodec2.pyx' doesn't match any files

Error installing sbapp from spec 'sbapp[macos]'.

<details><summary>cat /Users/tst/.local/pipx/logs/cmd_2024-11-07_03.54.39_pip_errors.log</summary>
<p>
PIP STDOUT

Collecting sbapp[macos]
  Using cached sbapp-1.1.3-py3-none-any.whl.metadata (15 kB)
Collecting rns>=0.8.4 (from sbapp[macos])
  Using cached rns-0.8.5-py3-none-any.whl.metadata (20 kB)
Collecting lxmf>=0.5.7 (from sbapp[macos])
  Using cached lxmf-0.5.7-py3-none-any.whl.metadata (12 kB)
Collecting kivy>=2.3.0 (from sbapp[macos])
  Using cached Kivy-2.3.0.tar.gz (24.0 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting pillow>=10.2.0 (from sbapp[macos])
  Using cached pillow-11.0.0-cp313-cp313-macosx_10_13_x86_64.whl.metadata (9.1 kB)
Collecting qrcode (from sbapp[macos])
  Using cached qrcode-8.0-py3-none-any.whl.metadata (17 kB)
Collecting materialyoucolor>=2.0.7 (from sbapp[macos])
  Using cached materialyoucolor-2.0.9.tar.gz (249 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting ffpyplayer (from sbapp[macos])
  Using cached ffpyplayer-4.5.2-cp313-cp313-macosx_10_13_x86_64.whl.metadata (4.2 kB)
Collecting sh (from sbapp[macos])
  Using cached sh-2.1.0-py3-none-any.whl.metadata (3.4 kB)
Collecting numpy<=1.26.4 (from sbapp[macos])
  Using cached numpy-1.26.4-cp313-cp313-macosx_14_0_x86_64.whl
Collecting pycodec2 (from sbapp[macos])
  Using cached pycodec2-3.0.1.tar.gz (93 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'

PIP STDERR

WARNING: sbapp 1.1.3 does not provide the extra 'macos'
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [38 lines of output]
      Traceback (most recent call last):
        File "/Users/tst/.local/pipx/shared/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/Users/tst/.local/pipx/shared/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/tst/.local/pipx/shared/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 333, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 303, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 521, in run_setup
          super().run_setup(setup_script=setup_script)
          ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 319, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 29, in <module>
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/Cython/Build/Dependencies.py", line 1010, in cythonize
          module_list, module_metadata = create_extension_list(
                                         ~~~~~~~~~~~~~~~~~~~~~^
              module_list,
              ^^^^^^^^^^^^
          ...<4 lines>...
              language=language,
              ^^^^^^^^^^^^^^^^^^
              aliases=aliases)
              ^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/Cython/Build/Dependencies.py", line 845, in create_extension_list
          for file in nonempty(sorted(extended_iglob(filepattern)), "'%s' doesn't match any files" % filepattern):
                      ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-oxrdntbo/overlay/lib/python3.13/site-packages/Cython/Build/Dependencies.py", line 117, in nonempty
          raise ValueError(error_msg)
      ValueError: 'pycodec2/pycodec2.pyx' doesn't match any files
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
</p>
</details> 

This is same error as in https://github.com/markqvist/Sideband/issues/50 but on macOS

How to resolve, please help!

markqvist, 2024-11-08:
Recently, I changed the build recipe a bit, with the intention of having platform-specific dependencies handled automatically. It *shouldn't* be necessary to add the `[macos]` anymore, so could you please try with a plain `pipx install sbapp`? Please let me know if that works or not.

Also, if anyone knows a way to virtualise / remotely access a recent macOS version, I'd be interested to hear it. The very old macbook I used for building and testing Sideband gave up, and I currently don't have access to any hardware for building / testing.

DeVLaB4U, 2024-11-14:
Not working.

pipx install sbapp
Fatal error from pip prevented installation. Full pip output in file:
    /Users/tst/.local/pipx/logs/cmd_2024-11-14_12.47.23_pip_errors.log

pip seemed to fail to build package:
    pycodec2

Some possibly relevant errors from pip install:
    error: subprocess-exited-with-error
    ValueError: 'pycodec2/pycodec2.pyx' doesn't match any files

Error installing sbapp.

`cat /Users/tst/.local/pipx/logs/cmd_2024-11-14_12.47.23_pip_errors.log`

```shell
PIP STDOUT
----------
Collecting sbapp
  Using cached sbapp-1.1.3-py3-none-any.whl.metadata (15 kB)
Collecting rns>=0.8.4 (from sbapp)
  Using cached rns-0.8.5-py3-none-any.whl.metadata (20 kB)
Collecting lxmf>=0.5.7 (from sbapp)
  Using cached lxmf-0.5.7-py3-none-any.whl.metadata (12 kB)
Collecting kivy>=2.3.0 (from sbapp)
  Using cached Kivy-2.3.0.tar.gz (24.0 MB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting pillow>=10.2.0 (from sbapp)
  Using cached pillow-11.0.0-cp313-cp313-macosx_10_13_x86_64.whl.metadata (9.1 kB)
Collecting qrcode (from sbapp)
  Using cached qrcode-8.0-py3-none-any.whl.metadata (17 kB)
Collecting materialyoucolor>=2.0.7 (from sbapp)
  Using cached materialyoucolor-2.0.9.tar.gz (249 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting ffpyplayer (from sbapp)
  Using cached ffpyplayer-4.5.2-cp313-cp313-macosx_10_13_x86_64.whl.metadata (4.2 kB)
Collecting sh (from sbapp)
  Using cached sh-2.1.0-py3-none-any.whl.metadata (3.4 kB)
Collecting numpy<=1.26.4 (from sbapp)
  Using cached numpy-1.26.4-cp313-cp313-macosx_14_0_x86_64.whl
Collecting pycodec2 (from sbapp)
  Using cached pycodec2-3.0.1.tar.gz (93 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'

PIP STDERR
----------
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [38 lines of output]
      Traceback (most recent call last):
        File "/Users/tst/.local/pipx/shared/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
[... 21 lines cut ...]
              module_list,
              ^^^^^^^^^^^^
          ...<4 lines>...
              language=language,
              ^^^^^^^^^^^^^^^^^^
              aliases=aliases)
              ^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-twiw6cht/overlay/lib/python3.13/site-packages/Cython/Build/Dependencies.py", line 845, in create_extension_list
          for file in nonempty(sorted(extended_iglob(filepattern)), "'%s' doesn't match any files" % filepattern):
                      ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/md/dkww984118x8ctmwnqk5wxt40000gn/T/pip-build-env-twiw6cht/overlay/lib/python3.13/site-packages/Cython/Build/Dependencies.py", line 117, in nonempty
          raise ValueError(error_msg)
      ValueError: 'pycodec2/pycodec2.pyx' doesn't match any files
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

```

DeVLaB4U, 2024-11-14:
@markqvist  ... a way to virtualise ...
https://github.com/quickemu-project/quickemu
On reddit somebody wrote "Quickemu works awesomely!"

markqvist, 2024-11-25:
Thanks for all the info! After quite a bit of wrangling, I was able to get macOS running in Quickemu, and produce a new universal build of Sideband.

The only problem now is, that since Quickemu doesn't support any kind of graphics acceleration, I can't actually run the application with the user interface displayed (I can redirect all GL calls to null, though, which gives me a seemingly functional Sideband app, albeit with an empty window).

If possible, can you please test [this new build of Sideband](https://github.com/markqvist/Sideband/releases/download/1.1.3/Sideband_1.1.4_universal.dmg) on your macOS machine(s)? If you have the possibility to test it on both intel and ARM hardware, that would be much appreciated!

DeVLaB4U, 2024-12-02:
https://github.com/markqvist/Sideband/releases/download/1.1.3/Sideband_1.1.4_universal.dmg
`Sorry, may be is late but " 404: Not Found."`
Can't download and test.