markqvist/Sideband #62: Sideband immediately quits at launch with no error message

xa4hf8, 2024-12-15:
**M1 Mac**
Mac OS Sequoia 15.1.1
Python 3.12.1
Sideband 1.2

[admin console log](https://github.com/user-attachments/files/18139056/Sideband.silently.exits.at.launch.txt)

[assistant analysis](https://chatgpt.com/share/675e561c-c970-8012-b554-e61f990eff28)

markqvist, 2024-12-15:
Is this running from `Sideband.app` provided in the dmg file, or installed via `pip`?

markqvist, 2024-12-15:
If it is running from the app package, could you try launching it via the terminal, and see if any additional error messages are output? Open a terminal, `cd` to the directory that the `Sideband.app` package is stored in, and run:

`./Sideband.app/Contents/MacOS/Sideband`

markqvist, 2024-12-15:
Ok, to investigate, I just launched a fresh macOS Sequoia instance running on Mac M1 hardware via EC2, and downloaded and installed the latest Sideband release. Everything runs as intended.

It could be that you simply have errors in your Reticulum configuration file, that prevents startup. Or a bit more unlikely, that you've upgraded from a *very* old version of Sideband, and the new version can't properly migrate database version and configuration files from that old version.

In that case, you can try moving the existing Sideband configuration folder (`~/.config/sideband`) to a backup location, which will make Sideband re-initialise config and database files on startup. If that works, you will have to manually import your old encryption key into the new config folder to use your existing address.

Moving this to the forums, since it currently does not appear to be an issue.