markqvist/Sideband #64: ImportError running Sideband on Raspberry Pi

KenSamson, 2024-12-16:
**Describe the Bug**
Error Launching sideband on Raspberry PI  

Error:   ImportError: cannot import name 'pn_announce_data_is_valid' from 'LXMF' (unknown location)

using latest RNS, and latest sideband.    RNSD by itself launches fine.

**To Reproduce**
(general) radiomanalpha@raspberrypi:~ $ rnsd --version
rnsd 0.8.8
(general) radiomanalpha@raspberrypi:~ $ 

(general) radiomanalpha@raspberrypi:~ $ sideband --version
sideband 1.2.0

(general) radiomanalpha@raspberrypi:~ $ sideband -v
[INFO   ] [Logger      ] Record log in /home/radiomanalpha/.kivy/logs/kivy_24-12-16_12.txt
[INFO   ] [Kivy        ] v2.3.0
[INFO   ] [Kivy        ] Installed at "/home/radiomanalpha/general/lib/python3.11/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.11.2 (main, Sep 14 2024, 03:00:30) [GCC 12.2.0]
[INFO   ] [Python      ] Interpreter at "/home/radiomanalpha/general/bin/python"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[2024-12-16 13:46:19] [Notice] Default scaling factor on this platform is unknown
[2024-12-16 13:46:19] [Notice] Looking for scaling info in /home/radiomanalpha/.config/sideband/app_storage/ui_scale
[INFO   ] [KivyMD      ] 1.2.0.dev0 (installed at "/home/radiomanalpha/general/lib/python3.11/site-packages/sbapp/kivymd/__init__.py")
[INFO   ] [Factory     ] 195 symbols loaded
[DEBUG  ] [Cache       ] register <kv.resourcefind> with limit=None, timeout=60
[DEBUG  ] [Cache       ] register <kv.image> with limit=None, timeout=60
[DEBUG  ] [Cache       ] register <kv.atlas> with limit=None, timeout=None
[INFO   ] [ImageLoaderFFPy] Using ffpyplayer 4.5.1
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil, img_ffpyplayer 
[DEBUG  ] [Cache       ] register <kv.texture> with limit=1000, timeout=60
[DEBUG  ] [Cache       ] register <kv.shader> with limit=1000, timeout=3600
[DEBUG  ] [Cache       ] register <kv.graphics.texture> with limit=None, timeout=None
[INFO   ] [Text        ] Provider: sdl2
[DEBUG  ] [Cache       ] register <kv.lang> with limit=None, timeout=None
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'3.1 Mesa 23.2.1-1~bpo12+rpt3'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'V3D 7.1'>
[INFO   ] [GL          ] OpenGL parsed version: 3, 1
[INFO   ] [GL          ] Shading version <b'1.40'>
[INFO   ] [GL          ] Texture max size <4096>
[INFO   ] [GL          ] Texture max units <24>
[DEBUG  ] [Shader      ] Fragment compiled successfully
[DEBUG  ] [Shader      ] Vertex compiled successfully
[DEBUG  ] [ImageSDL2   ] Load </home/radiomanalpha/general/lib/python3.11/site-packages/kivy/data/glsl/default.png>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Clipboard   ] Provider: xclip
[INFO   ] [CutBuffer   ] cut buffer support enabled
[INFO   ] [SoundFFPy   ] Using ffpyplayer 4.5.1
[DEBUG  ] [Audio       ] register SoundFFPy
[DEBUG  ] [Audio       ] register SoundSDL2
[DEBUG  ] [Audio       ] register MusicSDL2
[INFO   ] [Audio       ] Providers: audio_ffpyplayer, audio_sdl2 
 Traceback (most recent call last):
   File "/home/radiomanalpha/general/bin/sideband", line 5, in <module>
     from sbapp import main
   File "/home/radiomanalpha/general/lib/python3.11/site-packages/sbapp/main.py", line 261, in <module>
     from .sideband.core import SidebandCore
   File "/home/radiomanalpha/general/lib/python3.11/site-packages/sbapp/sideband/core.py", line 13, in <module>
     from LXMF import pn_announce_data_is_valid
 ImportError: cannot import name 'pn_announce_data_is_valid' from 'LXMF' (unknown location)

**Logs & Screenshots**
Please include any relevant log output. If applicable, also add screenshots to help explain your problem.

KenSamson, 2024-12-16:
[reticulum.confg.txt](https://github.com/user-attachments/files/18156608/reticulum.confg.txt)

Adding the reticulum config file...

markqvist, 2024-12-16:
What is the output of `lxmd --version`?

KenSamson, 2024-12-16:
It was 0.5.8... however...   I uninstalled it and re-installed it and bingo, nomadnet now works..

Thanks for the clue.. please close.

markqvist, 2024-12-16:
All good then, thanks.

closed 2024-12-16 by markqvist