markqvist/Sideband #65: latest gps fix may have broken the android repository server.

michael23505, 2024-12-17:
**Describe the Bug**
android repository server will not start

**To Reproduce**
attempt to start the repository server on android in sideband version 1.2.0

**Expected Behavior**
a host ip address and port should show on the repository server page after clicking start repository server

**Logs & Screenshots**
Please include any relevant log output. If applicable, also add screenshots to help explain your problem.

**System Information**
samsung galaxy a52 5g with android 14 and sideband version 1.2.0.

markqvist, 2024-12-17:
Any error messages show up in the log?

closed 2024-12-17 by markqvist

markqvist, 2024-12-17:
Thanks for reporting this. I've fixed it, and uploaded a new Android APK file to the `1.2.0` release. It is available now.

Could you please try to download it, and verify that it works on your devices?

michael23505, 2024-12-17:
It works. Thanks again.

simplyequipped, 2025-01-20:
I'm having this issue again on Sideband 1.30

markqvist, 2025-01-22:
Can you post the log output with debug logging turned on?

simplyequipped, 2025-01-22:
Welp, just started up Sideband again to check the log output and it's working fine. Nothing in the log related to the repository server.