markqvist/Sideband #66: Problem with background service on Android

michael23505, 2024-12-28:
Got a message that there may be a bug in the background service, but haven't been able to reproduce. Sideband on android does not stay active in the background. It stops after a few minutes of inactivity. Using several different phones and android version 14.

markqvist, 2024-12-28:
This *could* be due to power management killing the background process (even though it asks the OS nicely to be kept awake).

Have you excluded it from battery optimisations and/or allowed unrestricted background activity for the app? The settings UI for this varies a bit from vendor to vendor, but should be relatively straightforward to find.

Just a side-note. On Android versions where the API allows it, Sideband will request user permission to run unrestricted in the background on the first start, but if your particular version of Android does not support that, you will need to do it manually.

michael23505, 2025-01-01:
After some more testing, i've only gotten that message once or twice, but everything seems to be working properly. Thanks.

markqvist, 2025-01-27:
Closing due to inactivity and inability to reproduce.

closed 2025-01-27 by markqvist