markqvist/Sideband #69: Raspberry load.

k1e2n3t4, 2025-01-01:
**Read the Contribution Guidelines**
Before creating a bug report on this issue tracker, you **must** read the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md). Issues that do not follow the contribution guidelines **will be deleted without comment**.

- The issue tracker is used by developers of this project. **Do not use it to ask general questions, or for support requests**.
- Ideas and feature requests can be made on the [Discussions](https://github.com/markqvist/Reticulum/discussions). **Only** feature requests accepted by maintainers and developers are tracked and included on the issue tracker. **Do not post feature requests here**.
- After reading the [Contribution Guidelines](https://github.com/markqvist/Reticulum/blob/master/Contributing.md), delete this section from your bug report.

**Describe the Bug**
A clear and concise description of what the bug is.

**To Reproduce**
Describe in detail how to reproduce the bug.

**Expected Behavior**
A clear and concise description of what you expected to happen.

**Logs & Screenshots**
Please include any relevant log output. If applicable, also add screenshots to help explain your problem.

**System Information**
- OS and version
- Python version
- Program version

**Additional context**
Add any other context about the problem here.

k1e2n3t4, 2025-01-01:
Followed the tutorial on this site 4 times.  Each time with a new load on the Raspberry pi.   Sideband will not load or run.  
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-                                                                                                                                                             11-25) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jan  1 13:15:09 2025
admin@raspberrypi:~ $ sideband
[INFO   ] [Logger      ] Record log in /home/admin/.kivy/logs/kivy_25-01-01_1.tx                                                                                                                                                             t
[INFO   ] [Kivy        ] v2.3.1
[INFO   ] [Kivy        ] Installed at "/home/admin/.local/lib/python3.11/site-pa                                                                                                                                                             ckages/kivy/__init__.py"
[INFO   ] [Python      ] v3.11.2 (main, Sep 14 2024, 03:00:30) [GCC 12.2.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[2025-01-01 13:32:59] [Notice] Default scaling factor on this platform is unknow                                                                                                                                                             n
[2025-01-01 13:32:59] [Notice] Looking for scaling info in /home/admin/.config/s                                                                                                                                                             ideband/app_storage/ui_scale
Unable to connect to X server
[2025-01-01 13:33:07] [Notice] Launching platform-specific service for RNS and LXMF
[2025-01-01 13:33:07] [Notice] Purging stale messages... /home/admin/.config/sideband/app_storage/sideband.db
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-                                                                                                                                                             11-25) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jan  1 13:15:09 2025
admin@raspberrypi:~ $ sideband
[INFO   ] [Logger      ] Record log in /home/admin/.kivy/logs/kivy_25-01-01_1.tx                                                                                                                                                             t
[INFO   ] [Kivy        ] v2.3.1
[INFO   ] [Kivy        ] Installed at "/home/admin/.local/lib/python3.11/site-pa                                                                                                                                                             ckages/kivy/__init__.py"
[INFO   ] [Python      ] v3.11.2 (main, Sep 14 2024, 03:00:30) [GCC 12.2.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[2025-01-01 13:32:59] [Notice] Default scaling factor on this platform is unknow                                                                                                                                                             n
[2025-01-01 13:32:59] [Notice] Looking for scaling info in /home/admin/.config/s                                                                                                                                                             ideband/app_storage/ui_scale
Unable to connect to X server
[2025-01-01 13:33:07] [Notice] Launching platform-specific service for RNS and LXMF
[2025-01-01 13:33:07] [Notice] Purging stale messages... /home/admin/.config/sideband/app_storage/sideband.db

markqvist, 2025-01-02:
First, please format your issues according to the issue template, instead of just creating an issue that is literally a verbatim copy of the issue template ;)

That being said, according to this:

`Unable to connect to X server`

It looks like you are not running a desktop environment. Launching Sideband in GUI mode will of course not work without a desktop environment. If you want to run Sideband in headless mode on a server, use the daemon option `sideband -d` or `sideband --daemon`.

I'm closing this for now, as it looks more like a support request than an actual bug or issue with Sideband.

But feel free to add any additional details if you think this is actually a bug.

closed 2025-01-02 by markqvist

k1e2n3t4, 2025-01-02:
Where do I go for simple support?  Google only takes me this this page.   Running sideband from the desktop, brings up a black box with the header of sideband,  but nothing else.   no options nothing but a black box.