markqvist/Sideband #79: [Linux] Unhandled exception when opening Connectivity page

jaykayenn, 2025-05-18:
**Describe the Bug**
After another client (MeshChat) has been installed, unable to open Connectivity page on Sideband. App freezes with blank page. Other Sideband functionality works otherwise. 

**To Reproduce**
- Install, run, and close Sideband
- Install, run MeshChat
- Run Sideband again (with or without MeshChat running)

**Expected Behavior**
Continue working with shared instance with other apps. 

**Logs & Screenshots**
[INFO   ] [Logger      ] Record log in xxx/.kivy/logs/kivy_25-05-18_6.txt
[INFO   ] [Kivy        ] v2.3.1
[INFO   ] [Kivy        ] Installed at "xxx/.local/share/pipx/venvs/sbapp/lib/python3.12/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0]
[INFO   ] [Python      ] Interpreter at "xxx/.local/share/pipx/venvs/sbapp/bin/python"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[2025-05-18 20:29:46] [Notice]   Default scaling factor on this platform is unknown
[2025-05-18 20:29:46] [Notice]   Looking for scaling info in xxx/.config/sideband/app_storage/ui_scale
[2025-05-18 20:29:47] [Notice]   Launching platform-specific service for RNS and LXMF
[2025-05-18 20:30:05] [Error]    An unhandled <class 'AttributeError'> exception occurred: 'super' object has no attribute '__getattr__'
[2025-05-18 20:30:05] [Error]    Traceback (most recent call last):
  File "kivy/properties.pyx", line 961, in kivy.properties.ObservableDict.__getattr__
KeyError: 'connectivity_shared_access_label'

**System Information**
- Linux Mint 22.1 Cinnamon (focal)
- Python version: 3.12.3
- Program version: sbapp 1.6.1 (pipx)

**Additional context**
Reinstalling Sideband didn't help.

markqvist, 2025-05-18:
Thank you very much, that's fixed.

closed 2025-05-18 by markqvist

markqvist, 2025-05-26:
A new release hasn't been made since the bug was fixed. It will be included next time a release is published.

Ralith, 2025-07-02:
This seems to affect Windows, though I'm not sure how to find the logs to confirm it's the same error. On both Windows and Linux, I see this behavior without having ever installed any other client.

commit cd756239 (2025-05-18): Fixed incorrect widget ID. Closes #79.