markqvist/Sideband #80: Sideband 1.6.1 Android apk issues

r3k2, 2025-05-20:
Android apk
downloaded the latest version.

when I open the app after 30 secs or so I get a error MSG:
```
"The Reticulum and LXMF service seem to have disappeared, and Sideband is no longer connected.
This should not happens and probably indicates a bug in the background service. Please
restart Sideband to regain connectivity."
```
Ok I do that and it just keeps doing the same, I am new to reticulum, I been using nomadnet for 1 month or so and decided to try an android app and installed sideband 2 days a go, the latest apk you have on the release that seems very recent a couple days only. 
Sideband 1.6.1 on RNS v0.9.6 and LXMF v0.7.1

Hope this helps. 
PS: I made sure I give the app enough permissions to run on the background thinking that could be an issue.

markqvist, 2025-05-20:
Thanks for the info. If possible, could you enable debug logging in preferences, and see if anything shows up in Utilities -> Log Viewer after the error message appears?

markqvist, 2025-05-20:
Also the Android version and device type would be useful, if you don't mind sharing it.