markqvist/Sideband #81: LoRa transmission not working in Sideband 1.6.1 (reception works)

jokroese, 2025-05-24:
**Describe the Bug**

With the Sideband 1.6.1 release, I am able to receive LoRa transmissions but not send them. Other functionalities — including announcing streams, sending, and receiving over WiFi — work as expected. This suggests the issue is isolated to LoRa TX specifically.

**To Reproduce**

1. Configure Sideband 1.6.1 on a compatible system with both LoRa and WiFi interfaces.
2. Attempt to send message data over LoRa.
3. Observe that no LoRa transmissions occur.
4. Confirm that LoRa reception works correctly (e.g. receiving other nodes' traffic).
5. Confirm that both sending and receiving over WiFi work correctly.

**Expected Behavior**

Sideband should be able to both send and receive messages over LoRa, not just receive.

**System Information**

- **OS:** GrapheneOS
- **Sideband version:** 1.6.1

**Additional Context**

The issue does not occur over WiFi, indicating the stack and configuration are otherwise functional. LoRa reception confirms the hardware is working correctly. This appears to be a LoRa TX-specific issue.

y9Kap, 2025-07-18:
1. I used LilyGO LoRa T3S3
2. I used RNode version 1.82
3. I tried both ways to eliminate Bluetooth connection issues
4. BW was 125 and 62.5 (tried both at different times), CR&SF tried 6/8 and 7/7
5. Ok, im try now and write other comment
6. I can try to record a detailed video (I have three microcontrollers and three devices), but it will be in Russian, unfortunately, I don't know spoken English well. Or in very bad English. I'll make a disclaimer right away, I'm just a radio amateur and I don't do anything related to the military industry, because I love and support freedom of speech, and war and states are the opposite of freedom, unfortunately. I also apologize for the politics here, but at the present time I want normal technical and research interaction without prejudices due to politics, etc.

I also want to say that I checked it on the sideband version 1.5.0 and everything works there. Also a little later I can throw off the errors from the meshchat that pop up when trying to send a message from the new version of the sideband via RNode to Meshchat

y9Kap, 2025-07-18:
I tried with RNode 1.83 and the behavior of Sideband 1.7 was absolutely the same buggy. Then I tried RNode 1.83 with Sideband 1.5 and everything worked as usual. It seems that the problem is in the latest versions of Sideband

markqvist, 2025-07-18:
Hmm, very strange. Just to be sure, when you are testing, are all devices using a recent Sideband version? Because not so long ago, RNS changed from AES-128 to AES-256 encryption by default, so if you are using one version that only supports AES-128, and another version that only supports AES-256, that might be the problem. But if all are on the same version (and also any possible transport nodes in between), it should not be the problem. But since you are saying that old version works, it makes me think that maybe it has something to do with the AES-256 transition.

Strange thing is though, I cannot replicate this here at all, with any of the Android devices I have to test with, both on GrapheneOS and normal Android, with different Android versions from 9 to 15.

If you enable "Debug Logging" in Preferences, does anything come up in the debug log (found in "Utilities"). Also, if possible, you can enable debug logging and connect the phone to a computer via USB, enable Android developer mode and run `adb logcat | grep "python\|sidebandservice"` on the computer to get full detailed log output from Sideband and `rnsd` running on the phone. It may give hints to what is going on.

y9Kap, 2025-07-18:
Given: three Android devices with versions 15, 12, 12. All have Sideband version 1.7.0 (I downloaded it to one device and transferred it via Bluetooth to the other two). On the first device with Android 15 there is a connection to RNod version 1.82 via a serial port (type C-type C).The other two devices are connected to phones via Bluetooth to the RNode, which, in turn, are connected to power banks. Phone 2 can send a message to phone 1, but phone 1 cannot send a message to phone 2. I tried reinstalling the app, swapping RNodes, but my phone still couldn't send messages to phone 2 and 3.  

I'll try to do tests and get logs today. I also tried to write from sideband (1.7.0) to meshchat and there was an error, which I'll also post a bit later. And the error was related to encryption, if I understood correctly.

y9Kap, 2025-07-18:
[2025-07-18 21:01:53] [Error] Decryption failed on link <ac208b51d030b41db0a56d497a649fc5>. The contained exception was: Token HMAC was invalid

this exception was with try PTT,

[2025-07-18 21:05:31] [Error] Decryption with ratchets failed on <lxmf.delivery.27a655e3df21f08208153688f9721fab/cf924a605496062ae453dcf231545ec9>, reloading ratchets from storage and retrying

[2025-07-18 21:05:31] [Notice] Decryption succeeded after ratchet reload

this exception was when i try send text message from sideband to meshchat,

but announce worked successfully and without exception, i can see my devices (android and notebook with linux) in announce stream

y9Kap, 2025-07-18:
Yes, I ran all the tests again and realized that I am an inattentive and bad developer. We say: haste makes waste. Unfortunately, that's what happened.

I have an assumption that the problem was in the old version of RNode, that's why I couldn't communicate yesterday. Now I've reinstalled everything, everything works fine on two Androids, I'll attach a screenshot separately. Versions: RNode: 1.83, Sideband: 1.7.0, Meshchat: 2.1.0. LoRa device -- liligo t3s3

I'll also try to figure out how I can transfer cryptocurrency to you to compensate for the time spent on me. And thank you for advising me to update RNode! And for being so attentive to community messages, even if they are like mine...

y9Kap, 2025-07-18:
[image]

As promised, a screenshot of the program successfully working on the versions specified above

closed 2025-07-18 by markqvist